gphoto/libgphoto2 #917: Set camera configuration

martinsgronli, 2023-07-05:
**Bug description:**
Hi

I want to configure my camera with new values. However, using set_value() does not seem to cause a permanent change. I wonder if I use camera.set_config(config) in the wrong way (see below)?

**Camera:** 
Nikon Z7_2

**libgphoto2 and gphoto2 version:** 
2.5.27

**To reproduce:**
```
camera = gp.Camera()
camera.init()
config1 = camera.get_config()
iso1 = config.get_child_by_name("iso")
iso1.set_value("340") 
print(iso1.get_value())    # Print statement 1
camera.set_config(config1)

config2 = camera.get_config()
iso2 = config2.get_child_by_name("iso")
print(iso2.get_value())    # Print statement 2
```

msmeissn, 2023-07-05:
Can you try with the gphoto2 command line tool and see if there are errors reported?

for instance I do not think 340 is a valid ISO value.

320 would be a supported one.

martinsgronli, 2023-07-05:
You are absolutely right, I also used unsupported values for other configurations. Thank you.

msmeissn, 2023-07-05:
ok! :) good reminder to add error reporting to your python script :)

closed 2023-07-05 by msmeissn