gphoto/libgphoto2 #924: Canon EOS R100

thomaschaaf, 2023-09-04:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Canon EOS R100

**USB IDs**
Bus 001 Device 012: ID 04a9:3312 Canon, Inc. Canon Digital Camera

errors:

```
env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --summary

*** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-7: 'I/O problem') ***

root@prismabox-v2-86:/opt/mediauploader# cat my-logfile.txt
0.000010 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000041 main                        (2): gphoto2 2.5.28
0.000046 main                        (2): gphoto2 has been compiled with the following options:
0.000050 main                        (2):  + gcc (C compiler used)
0.000054 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000058 main                        (2):  + exif (for displaying EXIF information)
0.000061 main                        (2):  + no cdk (for accessing configuration options)
0.000065 main                        (2):  + no aa (for displaying live previews)
0.000068 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000072 main                        (2):  + readline (for easy navigation in the shell)
0.000079 main                        (2): libgphoto2 2.5.31
0.000083 main                        (2): libgphoto2 has been compiled with the following options:
0.000087 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen)
0.000091 main                        (2):  + gcc (C compiler used)
0.000095 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000098 main                        (2):  + EXIF (for special handling of EXIF files)
0.000105 main                        (2): libgphoto2_port 0.12.2
0.000109 main                        (2): libgphoto2_port has been compiled with the following options:
0.000112 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000116 main                        (2):  + gcc (C compiler used)
0.000119 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000123 main                        (2):  + EXIF (for vusb)
0.000127 main                        (2):  + USB (libusb1, for USB cameras)
0.000130 main                        (2):  + serial (for serial cameras)
0.000134 main                        (2):  + no resmgr (serial port access and locking)
0.000137 main                        (2):  + no ttylock (serial port locking)
0.000141 main                        (2):  + no lockdev (serial port locking)
0.000145 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000148 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000152 main                        (2): invoked with following arguments:
0.000156 main                        (2):   --debug
0.000159 main                        (2):   --debug-logfile=my-logfile.txt
0.000163 main                        (2):   --summary
0.000185 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.000227 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.000339 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000346 gp_port_init_localedir      (2): localedir has been set to /usr/local/share/locale (compile-time default)
0.000352 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.2'...
0.000416 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/disk'.
0.000637 gp_port_library_list        (2): found mtab fstype overlay
0.000652 gp_port_library_list        (2): found mtab fstype proc
0.000657 gp_port_library_list        (2): found mtab fstype sysfs
0.000661 gp_port_library_list        (2): found mtab fstype cgroup2
0.000665 gp_port_library_list        (2): found mtab fstype devtmpfs
0.000677 gp_port_library_list        (2): found mtab fstype devpts
0.000682 gp_port_library_list        (2): found mtab fstype tmpfs
0.000693 gp_port_library_list        (2): found mtab fstype hugetlbfs
0.000704 gp_port_library_list        (2): found mtab fstype mqueue
0.000714 gp_port_library_list        (2): found mtab fstype ext4
0.000723 gp_port_library_list        (2): found mtab fstype ext4
0.000732 gp_port_library_list        (2): found mtab fstype ext4
[... 132 lines cut ...]
0.043871 lookup_folder               (2): Found! / is 0x562a5ba1e690
0.043875 recurse_delete_folder       (2): Recurse delete folder 0x562a5ba1e690//
0.043878 delete_all_files            (2): Delete all files in folder 0x562a5ba1e690//
```

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

jensanjo, 2024-06-17:
@thomaschaaf did you try the R100 with the latest development release 2.5.31.1?
I am interested in using the R100.

jensanjo, 2024-07-16:
I tested the EOS R100 with libgphoto2 2.5.31.1, and it works all right. See attached files.

[autodetect.log](https://github.com/user-attachments/files/16254640/autodetect.log)
[list-all-config.txt](https://github.com/user-attachments/files/16254641/list-all-config.txt)
[summary.txt](https://github.com/user-attachments/files/16254642/summary.txt)

alfiolocastro, 2024-10-31:
where do I find 2.5.31.1 release? I could only find 2.5.31 (ubuntu noble) that doesn't recognize my R100

msmeissn, 2024-11-10:
(its currently not a release, its only in GIT)

thanks for verifying it works!

closed 2024-11-10 by msmeissn

BrainCrashSoft, 2024-12-11:
I'm having issues finding the 2.5.31.1 version... can someone provide a link?
I plan to get a EOS R100 but I need this to work first...