gphoto/libgphoto2 #927: You need to specify a folder starting with /store_xxxxxxxxx/

Jomy10, 2023-09-27:
**Describe the bug**

The error I am getting when copying a photo from the camera to the local file system is **You need to specify a folder starting with /store_xxxxxxxxx/**

The picture is coming from the standard path where gphoto2 saves the photo, so I don't know what to do about this.

#12 seems related

**Name the camera**
```
Model                          Poort
----------------------------------------------------------
Canon EOS 100D                 usb:020,012
```

**libgphoto2 and gphoto2 version**
```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs, clang, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1, clang, no ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
The following program gives me the error with following output:

```
Camera: 0x600002748210
Capturing...
File on camera: //capt0000.cr2
       file cfile_info reported flags: 133
    cfile_info reported mtime: 1695837784
     cfile_info reported size: 19638926
     cfile_info reported type: image/x-canon-raw
[gp_context_error] You need to specify a folder starting with /store_xxxxxxxxx/
[gp_camera_file_get [gphoto2-camera.c:1693]] 'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -1
Failed to get file
Deleting
CFile: '/'   'capt0000.cr2'
```

the program:
```c
#include <gphoto2/gphoto2.h>
#include <gphoto2/gphoto2-port.h>
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

static void log_error(GPLogLevel level, const char *domain, const char *str, void* data) {
  printf("[%s] %s\n", domain, str);
}

int main(int argc, char** argv) {
  Camera* camera;
  GPContext* context = gp_context_new();

  // port
  gp_log_add_func(GP_LOG_ERROR, log_error, NULL);
  
  gp_camera_new(&camera);

  int retval = gp_camera_init(camera, context);
  if (retval != GP_OK) {
    printf("Camera initialization failed with error code %i\n", retval);
    exit(1);
  }
  
  // Capture
  CameraFile* cfile;
  CameraFilePath cfile_path;
  CameraFileInfo cfile_info;

  printf("Capturing...\n");
  retval = gp_camera_capture(camera, GP_CAPTURE_IMAGE, &cfile_path, context);
  if (retval != GP_OK) {
    printf("Failed to capture\n");
    exit(1);
  }

  printf("File on camera: %s/%s\n", cfile_path.folder, cfile_path.name);

  retval = gp_camera_file_get_info(camera, cfile_path.folder, cfile_path.name, &cfile_info, context);
  printf ("       file cfile_info reported flags: %d\n", cfile_info.file.fields);
  if (cfile_info.file.fields & GP_FILE_INFO_MTIME) printf ("    cfile_info reported mtime: %ld\n", cfile_info.file.mtime);
  if (cfile_info.file.fields & GP_FILE_INFO_SIZE) printf ("     cfile_info reported size: %llu\n", cfile_info.file.size);
  if (cfile_info.file.fields & GP_FILE_INFO_TYPE) printf ("     cfile_info reported type: %s\n", cfile_info.file.type);

  int fd = open("file.cr2", O_CREAT | O_WRONLY, 0644);
  retval = gp_file_new_from_fd(&cfile, fd);
  if (retval != GP_OK) {
    printf("Failed to create file\n");
     goto CLEANUP;
  }
  retval = gp_camera_file_get(camera, cfile_path.folder, cfile_path.name, GP_FILE_TYPE_RAW, cfile, context);
  if (retval != GP_OK) {
    printf("Failed to get file\n");
    goto CLEANUP;
  }

  CLEANUP:
  gp_file_free(cfile);
  printf("Deleting\n");
  retval = gp_camera_file_delete(camera, cfile_path.folder, cfile_path.name, context);
  if (retval != GP_OK) {
    printf("Failed to delete file\n");
  }

  printf("CFile: '%s'   '%s'\n", cfile_path.folder, cfile_path.name);

  gp_camera_exit(camera, context);
}
```

msmeissn, 2023-09-27:
try it with GP_FILE_TYPE_NORMAL   (its a bit counter intuitive, but the CR2 files are "normal" files here)

Jomy10, 2023-09-27:
Ah, I see, thank you!

closed 2023-09-27 by Jomy10