gphoto/libgphoto2 #928: Modern Pentax Cameras not supported, possibly due to outdated PKRemote library, not PKTriggerTool library

ian-morgan99, 2023-09-27:
I am working with two solutions, which are both seeming dependent on LibGphoto2. 

One is 'Astroberry' which uses Indilib which uses LibGphoto2.
The other is Benro Polaris which as far as I understand, is also using a similar approach.

Neither of these solutions recognise modern Pentax cameras, such as the Pentax K3iii.
These solutions both fail to be able to use the LiveView feature of the Pentax K1ii.

As far as I can see, the Gphoto library is based on an old PKRemote project, which doesn't appear to be actively updated, and is missing features of a number of newer Pentax models. I believe this is why the K3iii for example, is completely unsupported.

There is a newer project, PKTriggerCord (https://github.com/asalamon74/pktriggercord) which appears to have support for new camera models, such as the Pentax K3iii and Pentax K1ii.

I have only attempted to connect using these compiled solutions, so have not been able to test in a way that allows me to generate the following:

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
**To Reproduce**
Steps to reproduce the behavior:
When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

I will attempt to do so. In the mean time, could you consider upgrading the solution to use the newer Pentax PKTriggerCord library?

msmeissn, 2023-09-28:
we also need the USB ids of the K-1 II

msmeissn, 2023-09-28:
commit 3118a779151d9b39b907a2b2817929011b0a9bd3

again, we still need the USB ids of the K3-III and K1-II

ian-morgan99, 2023-09-28:
I will see how I can find that out. I have both cameras here .

msmeissn, 2023-09-29:
run

lsusb

with the camera attached

msmeissn, 2023-09-29:
see pktriggercord issue. ids added to current libgphoto2 GIT.

(the ids are largely used for detection purposes)

ian-morgan99, 2023-09-29:
Many thanks for your quick response on this 👍 

As I'm new to this world, how does this update find it's way into the downstream projects that use it?
Should I now go chase IndiLib , Astroberry and the folks at Benro who are using this library, to get them to update from it, or are there other steps that need to happen first?

Sorry for the newbie questions

msmeissn, 2023-09-29:
i put fixes in libgphoto2. the next release is still some time out, as i just did a release some weeks ago.
as soon as its out, indi and others pick it up usually.

perhaps they even do git snapshots, not sure. you would need to ask them

ian-morgan99, 2023-09-29:
Many thanks.

closed 2023-10-04 by msmeissn

ian-morgan99, 2023-10-04:
It appears that the initial issue is not resolved , because of other problems in the PKtriggercord library . The Pentax K1ii and Kiii are not working when used directly , so fixing the issue downstream didn't resolve the issue . I will raise a fresh issue once I can see how that could possible be addressed .

ian-morgan99, 2025-01-26:
I have returned to this, and am attempting to port my own implementation for newer Pentax cameras.
If there is anyone who has patience to support a beginner with this process. I would greatly appreciate your support.