gphoto/libgphoto2 #931: Nikon multi-shot shutter release (similar to EOSRemoteRelease)

JacksonLegge, 2023-10-20:
Nikon cameras support a PTP operation "InitiateCaptureRecInMedia" (Operation Code 0x9207) that will take as many shots as the cameras set release/bracketing mode permits. However, there doesn't seem to be support for this. 

It would be great to use this to be able to take several bracketed shots faster than just using trigger capture several times.

msmeissn, 2023-10-21:
We support and call PTP_OC_NIKON_InitiateCaptureRecInMedia for several years already.

I think you might be able to do this with Burst shots, e.g. setting BurstNumber (prop 0x5018).
Have not tried it yet myself.

msmeissn, 2023-10-21:
try it.
- set mode to continuous slow or high shooting
- gphoto2 --set-config burstnumber=3 --capture-image-and-download

closed 2023-10-21 by msmeissn