gphoto/libgphoto2 #932: Fujifilm X-T3 Tethered Capture Freezes Camera

aeewhite, 2023-11-01:
**Describe the bug**

When running gphoto2 in `--capture-tethered`  mode, the camera becomes unresponsive and no PTP events are shown in the output. Pressing the shutter button does nothing, as well as changing any camera settings (ISO, shutter speed, etc.) via the hardware dials.

Other capture modes (like `--capture-image` and `--capture-image-and-download`) work successfully most of the time, but do fail occasionally with PTP errors such as `Error (-110: 'I/O in progress')`

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

```
Model                          Port                                            
----------------------------------------------------------
Fuji Fujifilm X-T3             usb:000,001    
```

Camera is running firmware v5.01

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs, clang, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1, clang, no ltdl, EXIF, USB, serial without locking
```

I'm making this bug report from a Mac running Apple Silicon, but also reproduces on a Raspberry Pi 4 with the same versions of gphoto2+libgphoto2

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Fuji X-T3 in "USB Tether Shooting Auto" mode
2. Start gphoto with `--capture-tethered`

Debug output from `sudo gphoto2 --capture-tethered --debug --debug-logfile=debug.log`
[debug.log](https://github.com/gphoto/libgphoto2/files/13222226/debug.log)

aeewhite, 2023-11-04:
Just saw  this is duplicate of https://github.com/gphoto/gphoto2/issues/570 so I'll close this issue and report there

closed 2023-11-04 by aeewhite

msmeissn, 2023-11-09:
i aded a fix in git for non working tether remote control