gphoto/libgphoto2 #937: v2.5.31 Sony (at least A7C aka ILCE-7C) not working anymore - cause: "has_sony_mode_300"

Bluebrain2000, 2023-11-23:
I noticed that with v2.5.31 the lib does not work with my Sony Alpha 7C (ILCE-7C) anymore.
The connection to the camera itself is okay (e.g. gets serial number), but that's it.
After looking through the changes from the previous release, v2.5.30, I noticed there was a new sub added to ptp.h, "has_sony_mode_300" (line 4686), which returns 1 for the camera model ILCE-7C.

In ptp.c line 4512 the return value of has_sony_mode_300 decides, if PTP_CNT_INIT (in usb.c) is called with parameter value 0x12c or 0x0c8.
Until v2.5.30 this was always 0x0c8 (commented in 2.5.31 with "older mode (200)")

So looks like some older camera models (at least ILCE-7C) were given a new init parameter that's invalid.

**If I comment out line 4691, so that has_sony_mode_300 returns 0 and parameter value 0x0c8 is used with PTP_CNT_INIT like it did prior to v2.5.31, everything works again!**
`if (!strcmp(params->deviceinfo.Model, "ILCE-7C")) return 1;`

gphoto2 --summary
**compiled with NOT working libgphoto2 v2.5.31:**
```
Camera summary:
Manufacturer: Sony Corporation
Model: ILCE-7C
  Version: 2.00
  Serial Number: 00000000000000003285233103798107
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats:
Display Formats: Association/Directory, JPEG, ARW, MPEG, Unknown(b982)

Device Capabilities:
        File Download, No File Deletion, File Upload
        No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: Daylight (4)
F-Number(0x5007): error 2002 on query.
Focus Mode(0x500a): error 2002 on query.
Exposure Metering Mode(0x500b): error 2002 on query.
Flash Mode(0x500c): error 2002 on query.
Exposure Program Mode(0x500e): error 2002 on query.
Exposure Bias Compensation(0x5010): error 2002 on query.
Still Capture Mode(0x5013): error 2002 on query.
DOC Compensation(0xd200): error 2002 on query.
DRangeOptimize(0xd201): error 2002 on query.
Image size(0xd203): error 2002 on query.
Shutter speed(0xd20d): error 2002 on query.
Property 0xd20e: error 2002 on query.
Color temperature(0xd20f): error 2002 on query.
CC Filter(0xd210): error 2002 on query.
Aspect Ratio(0xd211): error 2002 on query.
Focus status(0xd213): error 2002 on query.
Zoom(0xd214): error 2002 on query.
Objects in memory(0xd215): error 2002 on query.
Property 0xd217: error 2002 on query.
Battery Level(0xd218): error 2002 on query.
Picture Effect(0xd21b): error 2002 on query.
AB Filter(0xd21c): error 2002 on query.
Property 0xd21d: error 2002 on query.
ISO(0xd21e): error 2002 on query.
Property 0xd21f: error 2002 on query.
Property 0xd221: error 2002 on query.
Capture Target(0xd222): error 2002 on query.
Property 0xd223: error 2002 on query.
Property 0xd22a: error 2002 on query.
Focus Area(0xd22c): error 2002 on query.
Live View Setting Effect(0xd231): error 2002 on query.
Property 0xd235: error 2002 on query.
Property 0xd236: error 2002 on query.
Property 0xd23d: error 2002 on query.
Property 0xd23e: error 2002 on query.
Property 0xd23f: error 2002 on query.
[... 167 lines cut ...]
Property 0xd250:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd25b:(readwrite) (type=0x2) Enumeration [0,1] value: 0
Property 0xd25c:(readwrite) (type=0x6) Range [0 - 0, step 100] value: 0
Property 0xd25d:(readwrite) (type=0x6) 16777216
Property 0xd25f:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd260:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd265:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd266:(readwrite) (type=0x2) Enumeration [0,1] value: 1
Autofocus(0xd2c1):(readwrite) (type=0x4) Enumeration [1,2] value: 2
Capture(0xd2c2):(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2c3:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2c5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Still Image(0xd2c7):(readwrite) (type=0x4) Enumeration [1,2] value: 1
Movie(0xd2c8):(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2c9:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Near Far(0xd2d1):(readwrite) (type=0x3) Range [-7 - 7, step 1] value: 2
Property 0xd2d5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d6:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d7:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d8:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d9:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2dd:(readwrite) (type=0x1) Range [-1 - 1, step 1] value: 1
Property 0xd2de:(readwrite) (type=0x4) Enumeration [1,2] value: 2

```

msmeissn, 2023-11-24:
thanks for the report, merged your fix.

closed 2023-11-24 by msmeissn

romain-ker, 2024-03-22:
Same issue also with the ILCE-7RM5 (A7RV)
Also fixed by commenting the corresponding [line 4695](https://github.com/gphoto/libgphoto2/blob/846ae9f2199eb8bcbcb6d8bfe47fc961a033519a/camlibs/ptp2/ptp.h#L4695) in ptp.h before recompiling the lib
Before that, I was able to record a video but no picture and no control over parameters except white balanced.
Now, I can remotely change shutter speed, aperture, and a lot more as well as take pictures through gphoto2.

Until now, and based on comments, it seems that A7C, A7RIV and A7RV are concerned and should be removed from that list. The list may still increase in the future

Ryther, 2024-04-13:
Same issue with A7M4 (ILCE-7M4).
Now I try to compile while commenting the line out.
Had no way to try another version since I use gphoto with the indi drivers and it require the latest version as of now.

knro, 2024-04-14:
Right, multiple reports about this. Any side effects besides the manual focus @Ryther ?

Ryther, 2024-04-14:
In gphoto2, not one I'm aware of; capturing and downloading, the preview and video capturing are all apparently working correctly.
I have a problem in kstars, tho, where if I try to take multiple shots it fails every other one, when using the "Preview" button, and abort on the second shot while trying to get a sequence.
If I try to take multiple shots using gphoto2, they're taken and downloaded without problem, so I think it's more a problem of indi/kstars.

msmeissn, 2024-04-21:
something seems off with this line. I disabled the 2 cameras more

alexisdrakopoulos, 2024-06-04:
Has there been any update on this? Does libgphoto2 work with a7c & a7cii?

Bluebrain2000, 2024-06-05:
Yes, they work fine as long as you deactivate "mode_300" for them.

msmeissn, 2024-06-05:
i have  commented out 7C and 7CII was never listed there from mode 300.

just probably not yet in a release.

nagmat84, 2024-06-29:
I found this thread while doing a Google search. 

I have got a Sony Alpha 7C II. On the user level, I see an "unknown error 150" when I try to import media from the camera via MTP. (Different programs: KDE dolphin, Gwenview, ...).

Under the hood, all those programs use libgphoto2, version 2.5.31.

There are other reports that Sony Alpha 7C II is supposed to work fine, however those reports refer to older versions of libgphoto2. I could not downgrade to test those versions, because 2.5.31 is the only version included in the package manager of my Linux distro.

I am not sure if my problem is related to the issue at hand. If not, I can open another issue.

ertdfgcvb, 2024-09-02:
Apologies for asking here: what’s the best approach to currenty get **libgphoto2** working for an affected Sony alpha model?  

- compile the modified source or the older v2.5.30? That would be out of my scope/knowledge, let alone re-link the files in macOS. Homebrew unfortunately only allows to install the current version and not something like b`rew install libgphoto2@2.5.30`. Any pointers on how to compile and link would be appreciated! 

- wait for the new version? When could we expect it approximately? 

Thank you for any input (and obviously for the fantastic work on gphoto/libgphoto!

crispianm, 2024-10-03:
Having the same problem with the Sony ILCE-7SM3, resolved by commenting out the relevant line and recompiling. Could we also comment out the line from mode 300 in release or is a better solution possible? Thanks.

andrewyguo, 2024-10-16:
Has this been fixed in 2.5.31.1?

knro, 2024-10-16:
Latest GIT is still broken for many Sony users even with the commented lines.

knro, 2024-10-20:
I finally had access to a Sony camera today and did a git bisect and found the cause. Please check https://github.com/gphoto/libgphoto2/pull/1044

msmeissn, 2024-10-21:
please check the fix i did

knro, 2024-10-25:
Thanks, I have access to the camera again to test. But I wasn't sure what's the proper way to capture a 30 seconds bulb exposure from gphoto2? I tried to use this:

```gphoto2 --capture-image-and-download --bulb=30```

But I get 

> Sorry, the property 'Bulb Mode' / 0xd2c7 is currently read-only

Ryther, 2024-10-25:
Hello @knro,
when testing the problem with the Sony camera on Stellarmate, I've used:

`gphoto2 --no-keep -F 3 -I 2 --capture-image-and-download`

Where

> -F, --frames=COUNT                         Set number of frames to capture (default=infinite)
> -I, --interval=SECONDS                     Set capture interval in seconds

knro, 2024-11-03:
Thank you, that worked and turns out sometimes the camera just take a very long time to download so I adjusted the timeout in the INDI driver.