gphoto/libgphoto2 #942: gphoto2 not listing SD card files, showing internal folder on Sony Alpha A7R IV

MeIorito, 2023-12-07:
I am facing an issue using the ILCE-7RM4 camera. 
The camera has three settings: Mass Storage, MTP and PC Remote Control. When using Mass Storage and MTP gphoto2 sees the folders where the images are stored. When using the PC Control Mode it only sees a folder called "ILCE-7RM4". 

For what i want to do i need to be in PC Control Mode, i want to shoot and save the images to a Raspberry Pi automatically. 
So i can shoot, and the pictures are saved to the SD-card. But i cant download the images from the camera to the Pi because it sees the wrong folder.

The camera name:
Sony DSC-A7r IV

gphoto2 version:
2.5.27

**To Reproduce**
gphoto2 --list-files

This gives the following output:
There is no file in folder "/"

msmeissn, 2023-12-08:
does 
gphoto2 --capture-image-and-download 
work?

msmeissn, 2023-12-08:
there is a capturetarget that can be configured, if you want immediate download check that the setting is on SDRAM

MeIorito, 2023-12-21:
No gphoto2 --capture-image-and-download doesn't work, neither does setting the capture target. Feels like it is being recognized as a storage device., eventhough it is in remote PC controle mode.