gphoto/libgphoto2 #944: Sony Alpha 6100 in MTP mode

kjetilk, 2023-12-11:
**Name of the camera**

Sony Alpha 6100

In this issue, I have set it up in MTP mode.

**USB IDs**

```
Bus 005 Device 025: ID 054c:0d13 Sony Corp. ILCE-6100
```

**camera summary output**

[mtp-summary.txt](https://github.com/gphoto/libgphoto2/files/13634495/mtp-summary.txt)

**camera configuration output**

[mtp-config.txt](https://github.com/gphoto/libgphoto2/files/13634503/mtp-config.txt)

**test capture**

*  gphoto2 --capture-image-and-download

Does not work, but works in #943 .

*  gphoto2 --capture-preview

Does not work and times out with the following error in debug:
```
*** Error ***              
PTP Timeout

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-10: 'Timeout reading from or writing to the port') ***       
```

[Logfile for this here](https://github.com/gphoto/libgphoto2/files/13634549/my-logfile.txt)

kjetilk, 2023-12-11:
Uh-oh. The camera became unresponsive after testing this... I couldn't even turn it off, had to take the battery out.

msmeissn, 2023-12-11:
In mtp mode Sonys are not allowing capture. It should however not crash the camera.

the logfile you sent was on a second try or so, it has not protocol content, just hanging communication

kjetilk, 2023-12-11:
> In mtp mode Sonys are not allowing capture. It should however not crash the camera.

Yeah!

> the logfile you sent was on a second try or so, it has not protocol content, just hanging communication

Oh, yes, indeed. Should I open a separate issue on that?

msmeissn, 2023-12-11:
no its just a followup issue. attach it here please

kjetilk, 2023-12-11:
Huh, now the USB id changed...:

```
kjetil@owl:~/scratch/camera$ lsusb | grep -i sony
Bus 005 Device 033: ID 054c:0d13 Sony Corp. ILCE-6100
kjetil@owl:~/scratch/camera$ LANG=C gphoto2 --auto-detect 
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:005,033     
```

kjetilk, 2023-12-11:
Anyway, here's the hang:

```
kjetil@owl:~/scratch/camera$ env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-preview
*** Error (-10: 'Timeout reading from or writing to the port') ***             
```
with
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/13638229/my-logfile.txt)

msmeissn, 2023-12-22:
I think the capture preview in this mode just hangs / crashes the camera. 
Not much we can do here, if the ID is provided by the current libgpoto2 gphoto2 will know not to try capture preview.

kjetilk, 2023-12-22:
> I think the capture preview in this mode just hangs / crashes the camera. 

Oh, oops!

> Not much we can do here, if the ID is provided by the current libgpoto2 gphoto2 will know not to try capture preview.

That works for me, the camera does not support it in MTP mode anyway.