gphoto/libgphoto2 #945: Sony A7M3 Product ID issue

quexichen1, 2023-12-12:
The Sony A7M3 product id written for the following models is 0x096f, but when I recently recorded the product IDs of some old and new Sony models, I found that A7M3 does not have this product ID.
{"Sony:Alpha-A7III (Control)",  0x054c, 0x096f, PTP_CAP|PTP_CAP_PREVIEW},

The product ID of the a7m3 pc control mode that I obtained is the following one in the project.
{"Sony:Alpha-A7 III (PC Control)",	0x054c, 0x0c34, PTP_CAP|PTP_CAP_PREVIEW}, 

And after setting the USB connection option in some new Sony models such as ILCE-7RM5 to select when connecting, the option pops up after connecting the camera to the mac pro computer, but I do not select it. The product id read out at this time is 0x096f.

So I want to confirm that the a7m3 product ID in the project is wrong, or may the product ID be different in different batches or sales regions?

msmeissn, 2023-12-12:
We usually get user reports or similar to add those ids.
the 0x0c34 id was taken from another git repo,
0x96f came from https://github.com/gphoto/libgphoto2/issues/343

I would currently assume 0x096f to be the correct one.

Currently having an ID too much does not hurt.

quexichen1, 2023-12-12:
Thanks for your reply, I will try to contact the author of issue #343

closed 2023-12-14 by quexichen1