gphoto/libgphoto2 #946: Mostly non-functional with Sony ILCE-7C

BenLand100, 2023-12-30:
I use this camera for astrophotography, and ultimately ended up here after Kstars/Ekos identified a "lack of exposure widget" for this camera as being a gphoto2 error. After debugging with gphoto2 CLI a bit, it appears my (?) ILCE-7C doesn't have any significant functionality, despite being listed as supported by gphoto2.  Any idea how to get to a more capable state?

I assume this is not an issue with my camera, as the same cables+camera work fine under a Windows ecosystem also using Sony's PTP https://github.com/dougforpres/ASCOMSonyCameraDriver

Things to note: 
* Camera is in PC Remote mode (same as windows)
* -PC- emblem shows up on camera indicating it's connected (same as windows)
* Certain things do seem to partially work
* Mostly, core functionality with this camera is missing

Basic capture functionality errors out, camera does nothing:

```
benland100@blandtoo ~ $ gphoto2 --capture-image
ERROR: Could not capture image.                                                
ERROR: Could not capture.
```
Curiously, `--trigger-capture` works, as in the camera takes a picture successfully, but the command itself fails:

```
benland100@blandtoo ~ $ gphoto2 --trigger-capture
ERROR: Could not trigger capture.                                              
*** Error (-1: 'Unspecified error') ***  
```

There are not many config options showing, though I can adjust e.g. `whitebalance` with `--set-config` and  `--get-config` and these seem to take effect on the camera.

```
benland100@blandtoo ~ $ gphoto2 --list-all-config
/main/actions/movie                                                            
Label: Movie Capture
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/opcode
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/status/serialnumber
Label: Serial Number
Readonly: 1
Type: TEXT
Current: 00000000000000003285211106169435
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 1
Type: TEXT
Current: Sony Corporation
END
/main/status/cameramodel
Label: Camera Model
Readonly: 1
Type: TEXT
Current: ILCE-7C
END
/main/status/deviceversion
Label: Device Version
Readonly: 1
Type: TEXT
Current: 2.00
END
/main/status/vendorextension
Label: Vendor Extension
Readonly: 1
Type: TEXT
Current: Sony PTP Extensions
[... 189 lines cut ...]
Property 0xd2e4: error 2002 on query.
Property 0xd2e5: error 2002 on query.
Property 0xd2e6: error 2002 on query.
```

**Name the camera**

Sony ILCE-7C (Control)

**libgphoto2 and gphoto2 version**

```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         x86_64-pc-linux-gnu-gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs, x86_64-pc-linux-gnu-gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, x86_64-pc-linux-gnu-gcc, no ltdl, EXIF, USB, serial lockdev lockin
```

BenLand100, 2023-12-30:
Looks like the fix for #937, which is already in the repo, resolves this, it's just not in the most recent release yet.

msmeissn, 2024-01-01:
yes, will be in next release

closed 2024-01-01 by msmeissn