gphoto/libgphoto2 #949: OM5 support status?

ontheway16, 2024-01-05:
Hi, I am wondering if anyone tested the library with OM System's OM5 cameras. I could not find its' name in supported cams list, or any mentions here in issues.

msmeissn, 2024-01-07:
we had not reports so far. it might be using the same protocol as OM1, so if you have one you could try testing it?

msmeissn, 2024-01-08:
I would giveit a  50/50 chance that it would work already with the OM-1 code.

ledgitbe, 2024-02-04:
I have an OM5 and I'd love to test and give feedback but I'm not sure where to start. If there is a branch with OM-5 support using the already existing OM-1 code, I would be able to test it. If it doesn't work with the OM-1 code, maybe chances are it will work with Olympus EM-5 MK III code ?

Using gphoto 2.5.28 with my OM5 connected in MTP mode:
```
~ » gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:006,023  
`

ledgitbe, 2024-02-04:
@ontheway16 Indeed this seems to be the case. The camera only offers me these modes:
- storage
- MTP
- webcam

And according to https://software.omsystem.com/omcapture/en/ the OM5 has no USB tethering mode.

ontheway16, 2024-02-05:
@ledgitbe then it's a version of em5 mk3, unfortunately. We will hope for OM5 mk2.

ledgitbe, 2024-02-06:
gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log

[capture.log](https://github.com/gphoto/libgphoto2/files/14179446/capture.log)

mszyndel, 2026-11-04:
Did some work here https://github.com/gphoto/libgphoto2/pull/1348

Non-flagship cameras don't support tethering, so they are limited to MTP protocol. Should be enough to trigger capture, but I didn't have a chance to test that.