gphoto/libgphoto2 #950: `gp_camera_exit` returns `-1: Unspecified error` for Fujifilm X100V

ciaron, 2024-01-10:
**Describe the bug**

Function `gp_camera_exit` returns `-1: Unspecified error` for Fujifilm X100V

This bug manifested itself in my workflow originally via the `rapid-photo-downloader` tool. There is a bug report here: https://github.com/damonlynch/rapid-photo-downloader/issues/117

I was able to reproduce the original problem using with a C program linking only to libgphoto2 (see below)

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

```
Model                          Port                                            
----------------------------------------------------------
Fuji Fujifilm X100V            usb:001,014 
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.28         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

```

**To Reproduce**

Steps to reproduce the behavior:

Compile the following C program:

```
#include <stdio.h>
#include "gphoto2/gphoto2.h"
#include "gphoto2/gphoto2-version.h"

int test() {
    Camera* camera = NULL;
    CameraWidget* config = NULL;
    int error = GP_OK;
    printf("gp_camera_new\n");
    error = gp_camera_new(&camera);
    if (error) return error;
    printf("gp_camera_init\n");
    error = gp_camera_init(camera, NULL);
    if (error) return error;
    printf("get_camera_get_config\n");
    error = gp_camera_get_config(camera, &config, NULL);
    if (error) return error;
    printf("gp_camera_exit\n");
    error = gp_camera_exit(camera, NULL);
    if (error) return error;
    printf("gp_camera_unref\n");
    error = gp_camera_unref(camera); 
    return error;
}

int main() {
    printf("libgphoto2 version %s\n", *gp_library_version(0));
    int result = test();
    printf("result: %d: %s\n", result, gp_result_as_string(result));
    return 0;
}

```

with

`gcc test.c -lgphoto2`

In my case, I'm testing with a locally installed copy of libgphoto2:

`gcc test.c  -I/home/ciaron/opt/libgphoto2/2.5.28/include/gphoto2 -L/home/ciaron/opt/libgphoto2/2.5.28/lib -lgphoto2 -lgphoto2_port -lm`

but the error exists with the standard apt-installed version.

Output is:

```
libgphoto2 version 2.5.28
gp_camera_new
gp_camera_init
get_camera_get_config
gp_camera_exit
result: -1: Unspecified error
```
expected output:

```
libgphoto2 version 2.5.28
gp_camera_new
gp_camera_init
get_camera_get_config
gp_camera_exit
gp_camera_unref
result: 0: No error
```
**Workaround**

The error seems to have been introduced by the addition of the Fujifilm X100V model to the codebase at version 2.5.28.

By removing the relevant line (2605) in camlibs/ptp2/library.c (the FujiFilm X100V ID), and recompiling, I was able to run the test code without error.

Let me know if I can help with testing any patches etc.

msmeissn, 2024-01-13:
this happens in camera_exit encounters a protocol error or so.

can you reproduce this with gphoto2 and capture a debugtrace?

ciaron, 2024-01-13:
From the code it looks like `gphoto2 --reset` calls `gp_camera_exit`. I did this but it didn't trigger the error. (debug trace attached)
[debug.txt](https://github.com/gphoto/libgphoto2/files/13928915/debug.txt)

msmeissn, 2024-01-15:
well actually reset does not call the camera driver.
can you try this with --summary ?

ciaron, 2024-01-20:
Here is the output from `--summary`:

```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X100V
  Version: 3.01
  Serial Number: 593535383034210209CB0030218FC7
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: 
Display Formats: JPEG, JFIF

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: External Memory
	VolumeLabel: 000000000000000000000000000000000000000000000000
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 127865454592 (121942 MB)
	Free Space (Bytes): 127421251584 (121518 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
FocusPosition(0xd171): error 200a on query.
Property 0xd21c: error 200a on query.
Focus Point(0xd347): error 200a on query.
LensZoomPos(0xd170): error 200a on query.
BatteryLevel(0xd242): error 200a on query.
LiveViewImageSize(0xd174): error 200a on query.
VideoOutOnOff(0xd168): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
ForceMode(0xd230): error 200a on query.
Property 0xd16e: error 200a on query.
FocusArea1(0xd372): error 200a on query.
FaceDetectionMode(0xd020): error 200a on query.
RawCompression(0xd022): error 200a on query.
GrainEffect(0xd023): error 200a on query.
SetEyeAFMode(0xd024): error 200a on query.
FocusPoints(0xd025): error 200a on query.
MFAssistMode(0xd026): error 200a on query.
InterlockAEAFArea(0xd027): error 200a on query.
Shadowing(0xd029): error 200a on query.
CropMode(0xd16f): error 200a on query.
TNumber(0xd02f): error 200a on query.
FocusArea4(0xd395): error 200a on query.
HighLightTone(0xd320): error 200a on query.
ShadowTone(0xd321): error 200a on query.
LongExposureNR(0xd322): error 200a on query.
FullTimeManualFocus(0xd323): error 200a on query.
LensISSwitch(0xd346): error 200a on query.
InstantAFMode(0xd34a): error 200a on query.
PreAFMode(0xd34b): error 200a on query.
LMOMode(0xd34d): error 200a on query.
ISMode(0xd351): error 200a on query.
FocusCheckMode(0xd35e): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
FileNamePrefix1(0xd365): error 200a on query.
FileNamePrefix2(0xd366): error 200a on query.
FocusArea3(0xd374): error 200a on query.
TotalShotCount(0xd310): error 200a on query.
ExposurePreview(0xd359): error 200a on query.
FrameGuideGridInfo1(0xd375): error 200a on query.
FrameGuideGridInfo2(0xd376): error 200a on query.
CustomDispInfo(0xd36e): error 200a on query.
ViewMode1(0xd33f): error 200a on query.
CustomAutoPowerOff(0xd364): error 200a on query.
LockButtonMode(0xd34e): error 200a on query.
WideDynamicRange(0xd02e): error 200a on query.
LensNameAndSerial(0xd36d): error 200a on query.
LensUnknownData(0xd38a): error 200a on query.
BatteryInfo1(0xd36a): error 200a on query.
BatteryInfo2(0xd36b): error 200a on query.
FunctionLockCategory1(0xd36f): error 200a on query.
FunctionLockCategory2(0xd370): error 200a on query.
SensitivityFineTune1(0xd222): error 200a on query.
SensitivityFineTune2(0xd223): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
LensFNumberList(0xd38d): error 200a on query.
LensFocalLengthList(0xd38e): error 200a on query.
Property 0xd17b: error 200a on query.
```

hfiguiere, 2024-01-20:
Does this happen with ligphoto2 2.5.31 ?

ciaron, 2024-01-20:
> Does this happen with ligphoto2 2.5.31 ?

Yes it does.

ciaron, 2024-02-11:
attaching output from `gphoto2 --summary --debug`
[gphoto2-summary-output.txt](https://github.com/gphoto/libgphoto2/files/14232598/gphoto2-summary-output.txt)

ciaron, 2024-06-04:
I think your commit 5f9f6c8, plus the same change in `camera_unprepare_capture` (see below) works fixes the error.

```
index 80d528e24..1fdea073c 100644
--- a/camlibs/ptp2/config.c
+++ b/camlibs/ptp2/config.c
@@ -628,8 +628,10 @@ camera_unprepare_capture (Camera *camera, GPContext *context)
                                params->inliveview = 0;
                        }
 
-                       propval.u16 = 0x0001;
-                       C_PTP (ptp_setdevicepropvalue (params, PTP_DPC_FUJI_PriorityMode, &propval, PTP_DTC_UINT16));
+                       if (have_prop(camera, params->deviceinfo.VendorExtensionID, PTP_DPC_FUJI_PriorityMode)) {
+                               propval.u16 = 0x0001;
+                               C_PTP (ptp_setdevicepropvalue (params, PTP_DPC_FUJI_PriorityMode, &propval, PTP_DTC_UINT16));
+                       }
                        return GP_OK;
                }
                break;
```

I've been able to use this patched version with `gphoto2 -P` to download from the camera and with RapidPhotoDownloader to get images off the camera without problems.

The camera doesn't, as far as I can tell, support tethering so I haven't used any of the capture commands.

closed 2024-07-16 by ciaron