gphoto/libgphoto2 #954: Supported Cameras

nsmith1024, 2024-01-15:
At the bottom of your github page:

https://github.com/gphoto/libgphoto2

There is a "links" section, where you links to a "camera list", and "camera remote control list and doc".

1. Can someone please tell me the difference between those two lists?
2. Are these lists up-to-date?

Thanks

msmeissn, 2024-01-23:
camera list is a list that is autogenerated from / included in libgphoto2 itself. It is updated with the current release.
Note that for PTP cameras we detect also new PTP cameras "by class".

"camera remote control list and doc" is a manual edited table, it lacks a lot of newer cameras.

If there are conflicts I can go and and try resolve them, but currently do not have the time.

ClemSc, 2024-08-14:
Sorry for posting an information question here. I was looking in this list for the Canon EOS R50 and R100, and they are missing (the R10 is listed in the "camera list"). This means they are not supported, or not tested ?