gphoto/libgphoto2 #959: With gcc 14 around the corner, we really need to fix our compiler warnings

ndim, 2024-01-31:
BTW looks lile with latest glibc and gcc 14.x test suite build fails because it cannot be build wit `-D_FORTIFY_SOURCE=3` which turns `-Wimplicit-int` and `-Wimplicit-function-declaration` into errors
```console
/usr/bin/g++ -DHAVE_CONFIG_H -I. -I..  -I../libgphoto2_port -I..  -std=c++98 -pedantic -Wall -Wextra -Werror  -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -c -o test_pedantic_cxx98-test-pedantic-compilation.o `test -f 'test-pedantic-compilation.cxx' || echo './'`test-pedantic-compilation.cxx
test-pedantic-compilation.cxx: error: -Wno-implicit-int detected - is this intentional ? [-Werror]
test-pedantic-compilation.cxx: error: -Wno-implicit-function-declaration detected - is this intentional ? [-Werror]
cc1plus: all warnings being treated as errors
make[2]: *** [Makefile:1308: test_pedantic_cxx98-test-pedantic-compilation.o] Error 1
```

_Originally posted by @kloczek in https://github.com/gphoto/libgphoto2/issues/773#issuecomment-1917823555_

We need a systematic effort in eliminating compiler warnings, both for gcc 14 and for ourselves.

ndim, 2024-02-05:
I have set up a VM with Fedora 40 for a quick check, but I could not reproduce the original post.

I want to merge the two `configure` scripts into one first, and only then start fixing the warnings. That should save on build times.

kloczek, 2024-02-05:
Reported issue is result of use latest devel glibc with `-Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3` in CFLAGS which changes those warnings into errors.

Instead wasting time on create Fedora build env better would be just sort out those warnings.

ndim, 2024-02-05:
Not sure why trying to reproduce a reported problem would a waste of time...

Anyway, I could not reproduce the issue using the Fedora 40 packages

    gcc-14.0.1-0.4.fc40.x86_64
    glibc-2.38.9000-39.fc40.x86_64

running

    autoreconf -vis .
    configure --with-camlibs=everything
    make -j$(nproc) CFLAGS="-O2 -g -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3" CXXFLAGS="-O2 -g -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3" all check

on libgphoto2 commit 846ae9f2199eb8bcbcb6d8bfe47fc961a033519a.

What is "[the] latest devel glibc"? What libgphoto2 commit are you building? Using what commands?

ndim, 2024-02-07:
Another thing, this time building for i686:

    https://src.fedoraproject.org/rpms/libgphoto2/pull-request/5

yselkowitz, 2024-02-07:
> Another thing, this time building for i686. Quoting https://src.fedoraproject.org/rpms/libgphoto2/pull-request/5:
> 
> ```
> ptp2/chdk.c:1186:41: error: passing argument 3 of 'jpeg_mem_dest' from incompatible pointer type [-Wincompatible-pointer-types]
> /usr/include/jpeglib.h:922:43: note: expected 'long unsigned int *' but argument is of type 'uint64_t *' {aka 'long long unsigned int *'}
> ```

This is already fixed; see #941

msmeissn, 2024-02-09:
its not the -D_FORTIFY_SOURCE=3 usually turning on errors, but try 

-fhardened   or so which is new with gcc14

kloczek, 2024-02-09:
> its not the -D_FORTIFY_SOURCE=3 usually turning on errors, but try
> 
> -fhardened or so which is new with gcc14

I'm not using such option.

kloczek, 2024-02-09:
> is a downstream-only thing in Fedora, related to its use of annobin. If this reproduces in Fedora rawhide, it should be raised as an annobin bug.

Why not tolerating some exact warning is kind of bug? 🤔

OneRaichu, 2024-08-09:
you can try to use -fpermissive to avoid it.

msmeissn, 2024-10-02:
i think its done in head