gphoto/libgphoto2 #964: Fujifilm X100VI debug

ontheway16, 2024-02-22:
**Is your feature request related to a problem? Please describe.**
Fujifilm X100VI failed to capture.
**Describe the solution you'd like**
a gphoto2 version that supports X100VI
**Describe alternatives you've considered**
a gphoto2 version that supports X100VI
**Additional context**
summary and debug outputs added below, serials XX'ed.

Today I got a chance to test fujifilm X100 VI model, a 40MP cam with a non-changeable 23mm lens.
While the camera maker and model name retrieved succesfully, capture function appears not working. I also (succesfully) tested X-T5 and, its connection menu has some 'usb tethering' items, while this one not have such menus, yet. Saying yet since there will be a firmware update, according to dealer.

Adding the debug outputs below.
@msmeissn I have limited access to the cam, may be once again in a few days, I hope you may extract something useful from logs.
I have tried with .28, and also 28.1 versions of gphoto2.

[summary_X100VI.txt](https://github.com/gphoto/libgphoto2/files/14378695/summary_X100VI.txt)
[myerrorlog_X100VI.txt](https://github.com/gphoto/libgphoto2/files/14378696/myerrorlog_X100VI.txt)

ontheway16, 2024-02-23:
ok It seems X100 models has no tethering feature.

msmeissn, 2024-03-23:
i have added the id and the debug logfile, but i really concur with your guess it has no remote abilities.

closed 2024-03-23 by msmeissn