gphoto/libgphoto2 #966: Sony A7rIII (ILCE-7RM3A) ptp --trigger-capture leaves camera writing to memory card forever

thcipriani, 2024-03-19:
The sony a7riii in PC Control Mode can only take one image every two seconds. This is because writing to the memory card seems to requires the `ptp_getobject` function called from `camera_wait_for_event` inside `camlibs/ptp2/library.c`.

This means `gphoto2 --trigger-capture` triggers shutter release, but leaves the camera unable to perform any other commands until `gphoto2 --capture-image` is called (as it subsequently calls `ptp_getobject`). Capture image seems to be downloading all the images from the cameras sdram into my computer's memory (is that right?), so this limits how fast I can trigger photos.

 `gphoto2 --trigger-capture --wait-event=FILEADDED` also works, but is just as slow.

```
time gphoto2 --set-config shutterspeed=1/1000 --trigger-capture --wait-event=FILEADDED
Waiting for 'FILEADDED' event from camera. Press Ctrl-C to abort.              
UNKNOWN PTP Event c202, Param1 ffffc001
UNKNOWN PTP Property 5004 changed, "imagequality" to "RAW"
UNKNOWN PTP Property d21d changed, "PTP Property 0xd21d" to "0"
UNKNOWN PTP Event c201, Param1 ffffc001
FILEADDED capt0000.arw /                                                       
event found, stopping wait!

real    0m2.277s
user    0m0.059s
sys     0m0.085s
```

* * *
**Details**

Calling `gphoto2 --trigger-capture` triggers a shutter release, but then there's a memory card status indicator that shows 1 image writing to the memory card. Subsequent calls to `gphoto2 --trigger-capture` do nothing—they exit successfully without error, but no image capture is triggered, debug output for those calls looks fine—it's sending ptp commands to the camera successfully and thinks it's clicking the shutter.

The only way I've found to flush the image from the buffer to the memory card is by calling:  `gphoto2 --capture-image` (or `gphoto2 --wait-event=FILEADDED`)—this downloads the image to the card without taking a new image (debug indicates it's clicking the shutter—but the camera isn't doing anything).

The crux of this seems to be `ptp_getobject` function called from `camera_wait_for_event` inside `camlibs/ptp2/library.c`—if I return from `camera_wait_for_event` before `ptp_getobject`, the image is never flushed. If I return right after calling `ptp_getobject`: capture is written to the sdcard.

I worked around this (a bit) by setting continuous release capture mode:
```
gphoto2 --set-config /main/capturesettings/capturemode=29 # Choice: 29 Bracketing C 3.0 Steps 5 Pictures
gphoto2 --bulb 5 --capture-image
```

The camera lets you take a bunch of pictures, but then won't let you change camera settings or trigger another shutter release until it's done the camera→computer→camera dance.

* * *

**Name the camera**
```$ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Sony ILCE-7RM3A (PC Control)   usb:001,100
```

* * *

**libgphoto2 and gphoto2 version**
```$ gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.31.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

* * *

I haven't been able to find any other reports of this issue, so I thought I'd at least document it. Or perhaps there's some setting somewhere I missed. Please let me know if more details are helpful.

knro, 2024-03-19:
Not sure if it is a similar issue, but a user reported failure with Sony DSC-A7r IV (Control) using:
```
gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download
```
With this relevant failure:
```
3.310606 camera_sony_capture [library.c:4936](0): 'ptp_generic_getdevicepropdesc (params, PTP_DPC_CompressionSetting, &amp;dpd)' failed: PTP General Error (0x2002)
3.310620 gp_camera_capture [gphoto2-camera.c:1339](0): 'camera-&gt;functions-&gt;capture (camera, type, path, context)' failed: -1
ERROR: Could not capture image.

ERROR: Could not capture.
```

This is using latest libgphoto2 just compiled from master a couple of weeks ago. Full debug output attached. 

[gphoto.txt](https://github.com/gphoto/libgphoto2/files/14652419/gphoto.txt)

closed 2024-04-26 by thcipriani

msmeissn, 2024-04-27:
You can check out if there is a "capturetarget" setting. this would allow changing the capturetarget to the card on newer Sonys.

I think the A7C and AZR3a have it.

msmeissn, 2024-04-27:
For the compression setting perhaps open a new issue.

thcipriani, 2024-05-02:
> You can check out if there is a "capturetarget" setting. this would allow changing the capturetarget to the card on newer Sonys.
> 
> I think the A7C and AZR3a have it.

There is a `capturetarget` in the gphoto2 config for the a7riii and I have it set to the camera's card via `gphoto2 --set-config capturetarget=1` and it persists when I do `gphoto2 --get-config`

BUT, on the camera itself, under "menu" → "setup4" → "PC Remote Settings" → "Still Image Save Destination" there are only two options "PC Only" and "PC+Camera"

Some models appear to have "Camera Only"; however, in my case (where that option does not appear) it seems I have to read out the camera buffer BEFORE the camera will write to the card :(