gphoto/libgphoto2 #969: OM Systems OM-1 Mark II - not able to download

coldfire22x, 2024-03-27:
I'm able to capture and save to the SD card but not download the captured images.

I've built `libgphoto2` from [5f9f6c8](https://github.com/gphoto/libgphoto2/commit/5f9f6c87189d3223fb26d2db9a6a1a825bda9cf2) and `gphoto2` from [v2.5.8](https://github.com/gphoto/gphoto2/tree/v2.5.28). Not 100% sure if #967 should be enough for download to work?  No issues with my E-M1 Mark III's and I don't have an OM-1 Mark I to test with.

```
Olympus OM-1MarkII             usb:002,002
```

```
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.31.1       standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

```
$ gphoto2 --summary > gphoto2-summary.txt
$ gphoto2 --capture-image-and-download --debug --debug-logfile=capture-image-and-download.log
ERROR: Could not capture image.
ERROR: Could not capture.
```
[gphoto2-summary.txt](https://github.com/gphoto/libgphoto2/files/14769028/gphoto2-summary.txt)
[capture-image-and-download.log](https://github.com/gphoto/libgphoto2/files/14769030/capture-image-and-download.log)

coldfire22x, 2024-03-28:
I think I missed this yesterday because I always started my tests with some sort of gphoto2 command unrelated to capture (auto-detect, summary, etc.). It seems that capture and download DOES work, but only on the first attempt.

Toward the end of the first capture and download there is an error:

```
ptp_usb_getresp [usb.c:516] (0): PTP_OC 0x1016 receiving resp failed: PTP General Error (0x2002)
```

Then near the beginning of subsequent attempts, a different error:

```
ptp_usb_getresp [usb.c:516] (0): PTP_OC 0x1016 receiving resp failed: PTP Device Busy (0x2019)
```

Logs generated from the following commands are attached.

```
$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=capture1.log
New file is in location /store_00020001/DCIM/100OMSYS/OM120011.ORF on the camera
Saving file as OM120011.ORF
Deleting file /store_00020001/DCIM/100OMSYS/OM120011.ORF on the camera

$ ls OM120011.ORF
OM120011.ORF

$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=capture2.log
ERROR: Could not capture image.
ERROR: Could not capture.
```
[capture1.log](https://github.com/gphoto/libgphoto2/files/14782788/capture1.log)
[capture2.log](https://github.com/gphoto/libgphoto2/files/14782789/capture2.log)

msmeissn, 2024-04-27:
one thing  would be to try to comment out this line: (line 3267 in library.c)

                        CR (ptp_setdevicepropvalue (params, 0xD052, &propval, PTP_DTC_UINT16));

and see if that helps.

Also is this a dual image capture (raw+jpg)? if yes, try single-jpg or single-raw first if that works better?

coldfire22x, 2024-04-29:
It was set to single-raw .. I'll play with that setting on the camera with
and without that line you mentioned, thanks!

-Abe

coldfire22x, 2024-05-05:
@msmeissn Preliminary testing looks good on x86_64.  The first two logs are with [0697bb0](https://github.com/gphoto/libgphoto2/commit/0697bb0862f70a2f3a6930cee6a9f2433aeef609) (not working) and the last two are with the line you mentioned commented out.
[1-capture-image.log](https://github.com/gphoto/libgphoto2/files/15214733/1-capture-image.log)
[2-capture-image.log](https://github.com/gphoto/libgphoto2/files/15214734/2-capture-image.log)
[3-commented-out-capture-image.log](https://github.com/gphoto/libgphoto2/files/15214735/3-commented-out-capture-image.log)
[4-commented-out-capture-image.log](https://github.com/gphoto/libgphoto2/files/15214736/4-commented-out-capture-image.log)

coldfire22x, 2024-05-05:
Flipped the camera to RAW and plugged into an Orange Pi 5B. Looks like commenting out that line also works on aarch64:
[arm64-commented-out-capture-image-raw1.log](https://github.com/gphoto/libgphoto2/files/15214753/arm64-commented-out-capture-image-raw1.log)
[arm64-commented-out-capture-image-raw2.log](https://github.com/gphoto/libgphoto2/files/15214754/arm64-commented-out-capture-image-raw2.log)
[arm64-commented-out-capture-image-raw3.log](https://github.com/gphoto/libgphoto2/files/15214755/arm64-commented-out-capture-image-raw3.log)

coldfire22x, 2025-01-23:
Updated to the latest on `master` recently, and now I'm experiencing a seg fault when trying to use bulb. Not sure if this is specific to my camera model, or if it's a larger issue.

Built using 28958bc with the line mentioned above commented out:

```
abe@banana:~/src/libgphoto2$ gphoto2 --bulb=3 --capture-image --debug --debug-logfile=28958bc-bulb3.log
Bulb mode enabled (exposure time: 3s).
Segmentation fault (core dumped)
```

Built using 0697bb0 with the line mentioned above commented out:

```
abe@banana:~/src/libgphoto2$ gphoto2 --bulb=3 --capture-image --debug --debug-logfile=0697bb0-bulb3.log
Bulb mode enabled (exposure time: 3s).
New file is in location /store_00010001/DCIM/100OMSYS/P1230114.ORF on the camera

[0697bb0-bulb3.log](https://github.com/user-attachments/files/18526904/0697bb0-bulb3.log)
[28958bc-bulb3.log](https://github.com/user-attachments/files/18526903/28958bc-bulb3.log)

```

tenderlove, 2025-05-20:
Hi, I'm using the same camera and having similar problems. But commenting out the line [as mentioned](https://github.com/gphoto/libgphoto2/issues/969#issuecomment-2080430962) didn't seem to do the trick.

Applying this patch did fix it for me:

```diff
> git diff camlibs/ptp2/library.c
diff --git a/camlibs/ptp2/library.c b/camlibs/ptp2/library.c
index 7936fd28c..d76cf4aa8 100644
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -5554,6 +5554,7 @@ camera_olympus_omd_capture (Camera *camera, CameraCaptureType type, CameraFilePa
                        switch (event.Code) {
                        case PTP_EC_Olympus_ObjectAdded:
                        case PTP_EC_Olympus_ObjectAdded_New:    /* seen in newer traces, https://github.com/gphoto/gphoto2/issues/310 */
+                       case PTP_EC_Olympus_CaptureComplete:
                        case PTP_EC_ObjectAdded:
                                newobject = event.Param1;
                                goto downloadfile;
```

I noticed my log files have a lot of "unexpected unhandled event" messages coming from [here](https://github.com/gphoto/libgphoto2/blob/7bca13f074e82906ff5bc6cf67393f3eefd986bc/camlibs/ptp2/library.c#L5561).  I see `0xC103` which maps to `PTP_EC_Olympus_CaptureComplete`, but I'm seeing `0xc108` in @coldfire22x's logs.

With my above patch I'm able to do `--capture-image` as well as `--capture-image-and-download`.

I'm running with this commit of gphoto2: https://github.com/gphoto/gphoto2/commit/dfd3d4328d31a28436746191c3d4fa6b258ab797

and I'm linking against the commit in [this PR](https://github.com/gphoto/libgphoto2/pull/1123)

PR #1123 (2025-05-23): respond to CaptureComplete events from om-1 mkii cameras
Fixes #969 for me, but I'm not sure if this is the correct fix.