gphoto/libgphoto2 #970: Linking failure with clang: version script assignment of 'LIBGPHOTO2_INTERNAL' to symbol 'gpi_gphoto_port_type_map' failed: symbol not defined

NHOrus, 2024-03-27:
Attempts to build libgphoto2 with clang fails with error:
```
make[4]: Entering directory '/var/tmp/portage/media-libs/libgphoto2-2.5.30/work/libgphoto2-2.5.30-abi_x86_64.amd64/libgphoto2_port/libgphoto2_port'
/bin/sh ../libtool  --tag=CC   --mode=link clang  -O2 -pipe -march=native -Wall -Wmissing-declarations -Wmissing-prototypes -no-undefined -Wl,--version-script=/var/tmp/portage/media-libs/libgphoto2-2.5.30/work/libgphoto2-2.5.30/libgphoto2_port/libgphoto2_port/libgphoto2_port.ver -version-info 13:0:1 -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,--as-needed -o libgphoto2_port.la -rpath /usr/lib64 libgphoto2_port_la-gphoto2-port-info-list.lo libgphoto2_port_la-gphoto2-port-log.lo libgphoto2_port_la-gphoto2-port-version.lo libgphoto2_port_la-gphoto2-port.lo libgphoto2_port_la-gphoto2-port-portability.lo libgphoto2_port_la-gphoto2-port-result.lo -lltdl  -lpthread 
libtool: link: clang -shared  -fPIC -DPIC  .libs/libgphoto2_port_la-gphoto2-port-info-list.o .libs/libgphoto2_port_la-gphoto2-port-log.o .libs/libgphoto2_port_la-gphoto2-port-version.o .libs/libgphoto2_port_la-gphoto2-port.o .libs/libgphoto2_port_la-gphoto2-port-portability.o .libs/libgphoto2_port_la-gphoto2-port-result.o   -Wl,-rpath -Wl,//usr/lib64 -Wl,--as-needed //usr/lib64/libltdl.so -lpthread  -O2 -march=native -Wl,--version-script=/var/tmp/portage/media-libs/libgphoto2-2.5.30/work/libgphoto2-2.5.30/libgphoto2_port/libgphoto2_port/libgphoto2_port.ver -Wl,-O1 -Wl,-z -Wl,pack-relative-relocs   -Wl,-soname -Wl,libgphoto2_port.so.12 -o .libs/libgphoto2_port.so.12.1.0
ld.lld: error: version script assignment of 'LIBGPHOTO2_INTERNAL' to symbol 'gpi_gphoto_port_type_map' failed: symbol not defined
```

libgphoto2-2.5.30

NEWS said that gpi_gphoto_port_type_map was removed, but it apparently still exists in files
https://github.com/gphoto/libgphoto2/blob/5f9f6c87189d3223fb26d2db9a6a1a825bda9cf2/libgphoto2_port/libgphoto2_port/libgphoto2_port.ver#L76 
https://github.com/gphoto/libgphoto2/blob/5f9f6c87189d3223fb26d2db9a6a1a825bda9cf2/libgphoto2_port/gphoto2/gphoto2-port-info-list.h#L65

NHOrus, 2024-05-08:
diff with fix:
```
--- a/libgphoto2_port/libgphoto2_port/libgphoto2_port.ver       2024-03-28 06:27:36.576508646 +0000
+++ b/libgphoto2_port/libgphoto2_port/libgphoto2_port.ver       2024-03-28 06:28:09.394166837 +0000
@@ -71,7 +71,6 @@
 
 # These are only supposed to be used by libgphoto2 internally.
 LIBGPHOTO2_INTERNAL {
-       gpi_gphoto_port_type_map;
        gpi_enum_to_string;
        gpi_string_to_enum;
        gpi_string_to_flag;
--- a/libgphoto2_port/gphoto2/gphoto2-port-info-list.h  2024-03-28 06:27:36.577508636 +0000
+++ b/libgphoto2_port/gphoto2/gphoto2-port-info-list.h  2024-03-28 06:28:38.914859366 +0000
@@ -62,7 +62,6 @@
 
 #ifdef _GPHOTO2_INTERNAL_CODE
 #include <gphoto2/gphoto2-port-log.h>
-extern const StringFlagItem gpi_gphoto_port_type_map[];
 #endif
 
 int gp_port_info_new (GPPortInfo *info);
 ```

danfe, 2024-05-22:
It can be triggered with GCC/binutils as well by putting `-Wl,--no-undefined-version` on `CFLAGS`.  Yes, it would be nice to have this fixed.

closed 2024-05-25 by msmeissn