gphoto/libgphoto2 #972: GoPro Hero 10/Raspberry Pi: Error only in bookworm

outdoorbits, 2024-04-08:
Reading a GoPro seems to be not possible any more under bookworm. With the same hardware, the same thing works under bullseye. The results of "gphoto2 --version", "gphoto2 --auto-detect" and "gphoto2 --summary" are listed below. Except for "--auto-detect", all other commands tested under bookworm, such as "--get-all-files" also failed with the same error message.
I can rule out problems with the power supply (powered hub).
Thank you for your help!

##### bookworm @ Pi 4 and Pi 5 #####
> gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

> gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
GoPro HERO10 Black             usb:002,003     

> gphoto2 --port=usb:002,003 --summary
*** Error ***              
PTP General Error

*** Error ***              
An error occurred in the io-library ('Unspecified error'): No error description available
*** Error (-1: 'Unspecified error') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --port=usb:002,003 --summary

Please make sure there is sufficient quoting around the arguments.

> env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --port=usb:002,003 --summary
                                                                               
*** Error ***              
PTP General Error

*** Error ***              
An error occurred in the io-library ('Unspecified error'): No error description available
*** Error (-1: 'Unspecified error') ***       

##### bullseye @ Pi 4 #####
> gphoto2 --version
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

> gphoto2 --auto-detect 
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:002,003

> gphoto2 --port=usb:002,003 --summary 
Camera summary:                                                                
Manufacturer: GoPro
Model: HERO10 BLACK
  Version: H21.01.01.60.00
  Serial Number: ...

mijoost, 2024-07-31:
No progress yet? I'm having a similar/the same problem:

I don't have this issue with GoPro H 12 and libgphoto2-2.5.30 (which doesn't recognize that model).

I have this issue with libgphoto2-2.5.31 (which recognizes the H11/H12 model).
However, when specifying the option --camera='USB PTP Class Camera' to gphoto2, the problem does not show up. This camera model is what the earlier version defaulted to.

moll, 2024-12-31:
Just for the record, seeing the same here on Arch Linux with GPhoto2 v2.5.28, libgphoto2 v2.5.51, and GoPro HERO11 Black.