gphoto/libgphoto2 #974: Sony ILX-LR1 movie capture -> crash

nrplyx, 2024-04-16:
Trying to work with Sony ILX-LR1, device is connected via WiFi to PTPIP.
Majority of controls work properly, however when trying to start movie recording ON/OFF libgphoto2 crashes with segmentation fault. I'm using libgphoto2 ver 2.5.30, however 2.5.31 results are the same.
I tried to catch log-dump via gphoto2 interface:

#gphoto2 --set-config movie=1 --port=ptpip:192.168.122.1 --debug --debug-logfile=/tmp/www/lg
#do_page_fault(): sending SIGSEGV to gphoto2 for invalid write access to 00000000
#epc = 77b92e6b in ptp2.so[77b49000+a7000]
#ra  = 77b92ddf in ptp2.so[77b49000+a7000]
#Segmentation fault

Thanks for any suggestions!
[lg.txt](https://github.com/gphoto/libgphoto2/files/14988165/lg.txt)

nrplyx, 2024-04-19:
It looks like removing "*alreadyset = 1;" from _put_Sony_Movie function solves the problem for this camera:

static int
_put_Sony_Movie(CONFIG_PUT_ARGS)
{
	PTPParams *params = &(camera->pl->params);
	int val;
	PTPPropertyValue	value;
	GPContext *context = ((PTPData *) params->data)->context;
	CR (gp_widget_get_value(widget, &val));
	if (val)
		value.u16 = 2;
	else
		value.u16 = 1;

        C_PTP_REP (ptp_sony_setdevicecontrolvalueb (params, 0xD2C8, &value, PTP_DTC_UINT16 ));
	//*alreadyset = 1;
	return GP_OK;
}

msmeissn, 2024-04-19:
thanks, it was NULL in this property case.

closed 2024-04-19 by msmeissn