gphoto/libgphoto2 #976: Nikon Z6ii build in wifi - Access Denied

uwes-ufo, 2024-04-24:
This is not a bug.
I would like to use the integrated WiFi on my Nikon Z6_2. 

Unfortunately, the PTP_OC_GetObjectInfo (1008) command returns a PTP_RC_AccessDenied (200F). The camera clears AccessDenied as soon as I mark an image for upload and the PTP_OC_NIKON_AdvancedTransfer (9010) command is sent. However, command 9010 blocks if no image is selected to upload. 

Does anyone know a better way to disable "AccessDenied"?

msmeissn, 2024-04-26:
I know that NIKON_AdvancedTransfer exists, but its currently not hooked up in libgphoto2. I also checked the Z6_2 SDK, it does not list it as valid command. So I cant say what it does. It seems it was / is used by CoolPix Wifi.

is it not enabling browsing for all pictures via the regular PTP methods? Have not tested thsi on my Z6 for a while though.

uwes-ufo, 2024-04-26:
The access lock is active after the wifi connection is established. In both infrastructure and base station modes. (Not in smart device mode, however.) 

This means that it is initially not possible to browse through the file system. 

The PTP_OC_GetPartialObject (101b) command is also affected by the lock. However, the command PTP_OC_NIKON_GetPartialObjectEx (9431) does not. This could be a solution for downloading the images. Then there is 9361 in conjunction with PTP_OC_NIKON_GetObjectsMetaData (9434) as a poor replacement for 1008. But it doesn't work for images in the SDRAM. Finally, you would have to have a replacement for PTP_OC_GetThumb (100a): This could perhaps be 9530... 

All in all, it would be easier to solve the access block.

Vladimir-1212, 2026-09-13:
Hi! I ran into exactly the same problem while working on my own wireless receiver for the Nikon Z6 II, and after spending quite a bit of time comparing the camera behavior with Nikon Wireless Transmitter Utility, I think I finally figured out what is happening here.

The important part is that `AdvancedTransfer (0x9010)` does not seem to be an “unlock” command for `GetObjectInfo`.

It is actually part of Nikon's transfer queue / job protocol.

What I observed is roughly this:

```text
PTP/IP connection
        ↓
OpenSession
        ↓
AdvancedTransfer (0x9010)
        ↓
camera returns a descriptor for the next object
that is currently allowed to be transferred
        ↓
GetObjectInfo (0x1008)
        ↓
GetPartialObject (0x101B)
        ↓
GetPartialObject (0x101B)
        ↓
...
        ↓
file completed
        ↓
AdvancedTransfer (0x9010) with previous job status
        ↓
camera acknowledges the previous transfer
and returns the next transfer job
```

This also explains the `PTP_RC_AccessDenied (0x200F)` behavior.

The camera does not appear to allow arbitrary objects on the card to be accessed through this wireless transfer mode.

First, the object has to be issued by the camera through `AdvancedTransfer`.

Only after that does `GetObjectInfo (0x1008)` and then `GetPartialObject (0x101B)` work for that particular object.

So the logic is not:

```text
choose object → GetObjectInfo → call AdvancedTransfer to unlock it
```

but rather:

```text
AdvancedTransfer → camera gives you the allowed object → GetObjectInfo → download it
```

This was the key thing that confused me at first too.

I also checked what Nikon Wireless Transmitter Utility itself does. Interestingly, once the transfer job has been obtained, WTU does not seem to use some special Nikon-specific download operation. It downloads the actual file using the standard:

```text
GetPartialObject (0x101B)
```

in chunks.

So the proprietary/important part is mostly the transfer queue management through `0x9010`, not the actual file download.

Another detail I found: the next `AdvancedTransfer` call also acts as the status/acknowledgement for the previous job.

In my tests:

```text
0x2001 = previous transfer succeeded
0x2002 = previous transfer failed
```

So after finishing a file, I send another `AdvancedTransfer` with the previous job status, and the camera then provides the next transfer descriptor.

There was no separate `TransferComplete` command needed in the flow I observed.

This also explains why your test behaved differently after manually marking an image for transfer. Once an image exists in Nikon's transfer queue, `AdvancedTransfer` has a valid job to return, and access to that object becomes possible. Without a queued transfer job, the command can simply sit there waiting because there is nothing to deliver.

I verified this while implementing an Android receiver for the Z6 II and was able to get automatic RAW/JPEG transfers working with this flow.

So I don't think the solution is to disable `AccessDenied`. `AccessDenied` is basically Nikon telling us: “this object was not assigned to your current transfer job.”

Hope this helps — your issue actually pointed me in the right direction when I was trying to understand this protocol.