gphoto/libgphoto2 #978: Sony PTP Extensions

gabrielecoppi, 2024-06-10:
I discovered today, that Sony has opened up their PTP extension with this library [here](https://support.d-imaging.sony.co.jp/app/cameraremotecommand/en/index.html) is there any plan to include it?

hfiguiere, 2024-06-10:
It doesn't to be compatible with the licensing. The fact that you have to *request* it, and that it is even geography dependent is probably the first indication.

non-withstanding it's like binaries that would work on Linux.

msmeissn, 2024-06-10:
very cool. they published a PDF with PTP documentation too.

gabrielecoppi, 2024-06-10:
Yes, I think that document is the most important thing from the repo

kjetilk, 2024-06-14:
I looked through the license agreement. It is a terrible wall of text, but what's most unclear to me is whether there are actual software components that you'd need to implement the protocol and extensions, or if all you need is documentation. If it is only documentation, then surely, their copyright does not apply to an independent implementation?
It also mentions that there are open source components in there, in which case their additional restrictions may be in violation of open source licenses... It is an unnecessarily complex minefield because of all the legalese, but I'm not sure this would be something that couldn't be implemented because of it.

msmeissn, 2024-07-08:
see readme, Sony reached out to me and gave OK to use the documentation for our purposes, and also offered help.

zombodotcom, 2024-12-21:
has this been worked on recently?

watsonmw, 2024-12-23:
There's already been some work to support the newer Sony version of their protocol extensions (v3).  I plan to help with ILX / A9iii / A7R5 if I can.

getrav, 2025-02-01:
Any update on v2? 

| Operation Name                      | Hex Code   | Description                                                                                           |
|:------------------------------------|:-----------|:------------------------------------------------------------------------------------------------------|
| Name                                | Code       | Description                                                                                           |
| GetDeviceInfo                       | 0x1001     | same as PIMA 15740                                                                                    |
| OpenSession                         | 0x1002     | same as PIMA 15740                                                                                    |
| CloseSession                        | 0x1003     | same as PIMA 15740                                                                                    |
| GetObject                           | 0x1009     | same as PIMA 15740                                                                                    |
| GetThumb                            | 0x100A     | same as PIMA 15740                                                                                    |
| GetPartialObject                    | 0x101B     | same as PIMA 15740                                                                                    |
| SDIO_Connect                        | 0x9201     | This is for the authentication handshake                                                              |
| SDIO_GetExtDeviceInfo               | 0x9202     | Get the protocol version and the supported properties of the connected device                         |
| SDIO_SetExtDevicePropValue          | 0x9205     | Set a DevicePropValue for a device property                                                           |
| SDIO_ControlDevice                  | 0x9207     | Set the SDIControl value for the SDIControlCode                                                       |
| SDIO_GetAllExtDevicePropInfo        | 0x9209     | Obtain all support DevicePropDescs at one time                                                        |
|                                     |            | The host will send this operation at regular intervals to obtain the latest (current) camera settings |
| Compression Setting                 | 0x5004     | Get/Set the quality.                                                                                  |
| White Balance                       | 0x5005     | Get/Set the white balance.                                                                            |
| F-Number                            | 0x5007     | Get/Set the aperture value.                                                                           |
| Focus Mode                          | 0x500A     | Get/Set the focus mode.                                                                               |
| Exposure Metering Mode              | 0x500B     | Get/Set the exposure metering mode.                                                                   |
| Flash Mode                          | 0x500C     | Get/Set the ﬂash mode.                                                                                |
| Exposure Program Mode               | 0x500E     | Get/Set the exposure program mode.                                                                    |
| Exposure Bias Compensation          | 0x5010     | Get/Set the exposure bias compensation.                                                               |
| Still Capture Mode                  | 0x5013     | Get/Set the drive mode.                                                                               |
| Flash Compensation                  | 0xD200     | Get/Set the ﬂash compensation.                                                                        |
| Dynamic Range Optimizer             | 0xD201     | Get/Set the dynamic range optimizer.                                                                  |
| Image Size                          | 0xD203     | Get/Set the image size.                                                                               |
| Shutter Speed                       | 0xD20D     | Get/Set the shutter speed.                                                                            |
| Battery Level Indicator             | 0xD20E     | Get the battery level indicator.                                                                      |
| Color Temperature                   | 0xD20F     | Get/Set the color temperature.                                                                        |
| Biaxial Fine-Tuning G-M Direction   | 0xD210     | Get/Set the biaxial ﬁne-tuning G-M direction.                                                         |
| Aspect Ratio                        | 0xD211     | Get/Set the aspect ratio.                                                                             |
| Focus Indication                    | 0xD213     | Get the focus indication.                                                                             |
| Predicted Maximum File Size         | 0xD214     | Get the predicted maximum ﬁle size.                                                                   |
[... cut ...]
| Shutter Half-Release (S1) Button    | 0xD2C1     | nan                                                                                                   |
| Shutter Release (S2) Button         | 0xD2C2     | nan                                                                                                   |
| AEL Button                          | 0xD2C3     | nan                                                                                                   |
| AFL Button                          | 0xD2C4     | nan     /

msmeissn, 2025-02-02:
Most of v2/v3 stuff is supported already from before we got acccess.

what are you missing?

getrav, 2025-05-01:
Sorry I totally missed this. I couldn't find the ones that say NAN in libgphoto2

kjetilk, 2026-09-29:
One thing I'm missing is the ability to set the time and date from the command line. That's to keep the clock at high precision and in the UTC timezone. After chatting with various bots, it seems that my A6100 has a property 0xd25a that is likely time. But it doesn't seem like I can set that from gphoto2?

fliker09, 2026-09-29:
> One thing I'm missing is the ability to set the time and date from the command line. That's to keep the clock at high precision and in the UTC timezone. After chatting with various bots, it seems [...]

Afaik - it's hopeless. Even in the official docs from Sony nothing about it. Eventually developed this abomination to solve it to some degree:
https://github.com/fliker09/sony-location-info-link-sync