gphoto/libgphoto2 #980: Nikon Image Review

WilliamWelsh, 2024-06-18:
**Is your feature request related to a problem? Please describe.**
When my Nikon is NOT tethered and I take a picture, it shows it on the camera's screen. When it *is* tethered, it doesn't show anymore, and Image Review is grayed out in the settings. The only place I have seen this work, with a Nikon, is the CamRanger device.

**Describe the solution you'd like**
Maybe there is a ptp command or something that CamRanger is calling?

**Describe alternatives you've considered**
I thought maybe closing the ptp session would work, but it didn't. I'm not sure how CamRanger is doing it

WilliamWelsh, 2024-06-21:
Changing the application mode lets image review be enabled again, and the app still recognizes the camera, but image events are sent anymore. I'm able to get the storage ids, but if I try to get an object handles for a storage id it fails

closed 2024-06-21 by WilliamWelsh

liuding1983, 2024-07-24:
Same problem, how to solve it?

devintrue, 2024-07-24:
Same problem, how to solve it?

WilliamWelsh, 2024-07-24:
> Same problem, how to solve it?

Have to put the camera in Application mode, but to get image events in Application mode you have to call GetEvent yourself

devintrue, 2024-07-25:
> > Same problem, how to solve it?
> 
> Have to put the camera in Application mode, but to get image events in Application mode you have to call GetEvent yourself

Thanks. Unfortunately,  put the camera in Application mode  there are occasional errors when downloading photos “GP_ERROR_DIRECTORY_NOT_FOUND -107  The specified directory could not be found”