gphoto/libgphoto2 #983: seeking your kind assistance

letsfire, 2024-06-29:
```C
char* data;   
unsigned long int size = 1000;
CameraFile* file;
gp_file_new(&file);
CHECK_RESULT(gp_camera_file_get(camera, folder, name, GP_FILE_TYPE_NORMAL, file, context));
CHECK_RESULT(gp_file_get_data_and_size(file, &data, &size));
```

Why is the size correct, but the length of the data is only 10 bytes? I am use ptp2 (camlib) and libusb (port) in android with Canon EOS 60D

I/USB_CAMERA(11940): 5256827 - **the size of gp_file_get_data_and_size**
I/USB_CAMERA(11940): ����:\Exif  - **the data of gp_file_get_data_and_size**

I/USB_CAMERA(11940): Domain:gp_filesystem_get_file_impl, Data Len: 73
I/USB_CAMERA(11940): Domain:ptp, Data Len: 39
I/USB_CAMERA(11940): Domain:ptp, Data Len: 39
I/USB_CAMERA(11940): Domain:ptp, Data Len: 39
I/USB_CAMERA(11940): Domain:get_file_func, Data Len: 28
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 79
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2359364
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2354756
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 35
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 128
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 84
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2359364
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2354756
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 35
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 128
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 84
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2359364
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2354756
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 35
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 128
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 84
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2359364
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2354756
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 35
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 128
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 84
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2359364
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 43
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 2354756
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 35
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 128
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141
I/USB_CAMERA(11940): Domain:ptp_usb_sendreq, Data Len: 82
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 34
I/USB_CAMERA(11940): Domain:gp_port_write, Data Len: 199
I/USB_CAMERA(11940): Domain:ptp_usb_getdata, Data Len: 50
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 4670
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 41
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 57665
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 37
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 707
I/USB_CAMERA(11940): Domain:ptp_usb_getresp, Data Len: 54
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 39
I/USB_CAMERA(11940): Domain:gp_port_read, Data Len: 141

msmeissn, 2024-07-01:
> ```C
> char* data;
> unsigned long int size = 1000;
> CameraFile* file;
> gp_file_new(&file);
> CHECK_RESULT(gp_camera_file_get(camera, folder, name, GP_FILE_TYPE_NORMAL, file, context)); [...]

Where do you see the 10 bytes? The "����:\Exif" string?

The data returned here is in binary format, not a string, so do not use strlen or so
for displaying.

Ciao, Marcus
 
> I/USB_CAMERA(11940): 5256827 - **the size of gp_file_get_data_and_size**
> I/USB_CAMERA(11940): ����:\Exif  - **the data of gp_file_get_data_and_size**
>
> I/USB_CAMERA(11940): [...]

letsfire, 2024-07-01:
Thank you. With your reminder, I have resolved my issue.