gphoto/libgphoto2 #987: camera->functions->wait_for_event fails on Nikon Z9

Noxet, 2024-07-26:
**Describe the bug**

When calling the function `gp_camera_trigger_capture ` followed by a call to `gp_camera_wait_for_event` in order to download the image, the program sometimes crashes with the following message:

```
*** Contexterror ***              
PTP Device Prop Not Supported
'camera->functions->wait_for_event ( camera, timeout, eventtype, eventdata, context)' failed: -1
```

It happens sporadically and no apparent pattern is seen

**Name the camera**
Nikon Z9

**libgphoto2 and gphoto2 version**
libgphoto 2.5.30

**To Reproduce**
For example, use the example code in the repo: `sample-trigger-capture` which will call the function described above

Nanakbsk, 2024-08-21:
8

Noxet, 2024-08-22:
> 8

Very helpful comment :)

closed 2024-08-22 by Noxet

msmeissn, 2024-08-27:
(this is fixed in current GIT afaik)