gphoto/libgphoto2 #994: Please support XDG base directory specification

shmerl, 2024-08-26:
Currently on Linux libgphoto creates `$HOME/.gphoto` directory when used. Please use XDG base directory specification to avoid cluttering `$HOME`.

shmerl, 2024-08-26:
Looks like a dup of bug #249, so I'll close this.

shmerl, 2024-08-26:
Hm, on the other hand that's for the libary, so may be it needs a separate one. So let's keep it open.

ndim, 2024-09-05:
This is definitely an issue for the libgphoto2 library.

ndim, 2024-09-05:
So... the libgphoto2 code should, instead of just using `~/.gphoto`

  * Try `$XDG_CONFIG_HOME/gphoto`. If present, use it.
  * Otherwise, look for `~/.gphoto`.
      * If present, make sure the XDG based parent directories exist, creating directories as needed, then rename(2) `~/.gphoto` to `$XDG_CONFIG_HOME/gphoto`. Be aware of symlink attacks and similar things.
      * If not present, create `XDG_CONFIG_HOME/gphoto`. Be aware of symlink attacks and similar things.
  * Use `$XDG_CONFIG_HOME/gphoto`.

This should obviously apply on Linux, BSD.

Should it also apply on MacOS and Windows?

shmerl, 2024-09-05:
Not exactly. If `$XDG_CONFIG_HOME` is defined - then use `$XDG_CONFIG_HOME/gphoto` (create it if it doesn't exist).

If it's not defined use `$HOME/.config/gphoto` (create it if it doesn't exist).