gphoto/libgphoto2 #1014: gphoto2 --get-config exposurecompensation doesn't work with Canon EOS R7

sinardet, 2024-09-14:
Hi everyone,
I'm totaly new and would like to use the same command 
gphoto2 --get-config exposurecompensation , gphoto2 --set-config exposurecompensation doesn't work as well.

But when I try I've got only that below:
Label: Exposure Compensation
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 0
END
I've got all the rest of commands like :gphoto2 --get-config iso or aperture etc, they are working perfectly.
My camera is a Canon EOS R7.
Please let me know what can I do to fix it thanks so much
*******************************************

 gphoto2 --debug --get-config exibitioncompensation
0.000708 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000894 main                        (2): gphoto2 2.5.28
0.000973 main                        (2): gphoto2 has been compiled with the following options:
0.001041 main                        (2):  + gcc (C compiler used)
0.001103 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.001168 main                        (2):  + exif (for displaying EXIF information)
0.001235 main                        (2):  + no cdk (for accessing configuration options)
0.001298 main                        (2):  + no aa (for displaying live previews)
0.001361 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.001423 main                        (2):  + no readline (for easy navigation in the shell)
0.001494 main                        (2): libgphoto2 2.5.27
0.001566 main                        (2): libgphoto2 has been compiled with the following options:
0.001637 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.001704 main                        (2):  + gcc (C compiler used)
0.001842 main                        (2):  + ltdl (for portable loading of camlibs)
0.001906 main                        (2):  + EXIF (for special handling of EXIF files)
0.001977 main                        (2): libgphoto2_port 0.12.0
0.002050 main                        (2): libgphoto2_port has been compiled with the following options:
0.002117 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.002179 main                        (2):  + gcc (C compiler used)
0.002240 main                        (2):  + ltdl (for portable loading of iolibs)
0.002301 main                        (2):  + EXIF (for vusb)
0.002363 main                        (2):  + USB (libusb1, for USB cameras)
0.002423 main                        (2):  + serial (for serial cameras)
0.002484 main                        (2):  + no resmgr (serial port access and locking)
0.002545 main                        (2):  + no ttylock (serial port locking)
0.002607 main                        (2):  + no lockdev (serial port locking)
0.002670 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.002732 main                        (2): IOLIBS env var not set, using compile-time default instead
0.002794 main                        (2): invoked with following arguments:
0.002855 main                        (2):   --debug
0.002915 main                        (2):   --get-config
0.002976 main                        (2):   exibitioncompensation
0.003144 load_settings               (2): Creating gphoto config directory ('/home/kingsley/.gphoto')
0.003374 load_settings               (2): Loading settings from file '/home/kingsley/.gphoto/settings'.
0.003953 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.004056 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0'...
0.004546 foreach_func                (2): Called for filename '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0/disk'.
0.005680 gp_port_library_list        (2): found fstab fsname proc
0.005856 gp_port_library_list        (2): found fstab fsname PARTUUID=bd461752-01
0.005992 gp_port_library_list        (2): found fstab fsname PARTUUID=bd461752-02
0.006448 gp_port_library_list        (2): found mtab fsname /dev/root
0.006584 gp_port_library_list        (2): found mtab fsname devtmpfs
0.006672 gp_port_library_list        (2): found mtab fsname proc
0.006756 gp_port_library_list        (2): found mtab fsname sysfs
0.006843 gp_port_library_list        (2): found mtab fsname securityfs
0.006928 gp_port_library_list        (2): found mtab fsname tmpfs
0.007060 gp_port_library_list        (2): found mtab fsname devpts
0.007148 gp_port_library_list        (2): found mtab fsname tmpfs
0.007268 gp_port_library_list        (2): found mtab fsname tmpfs
0.007393 gp_port_library_list        (2): found mtab fsname cgroup2
0.007475 gp_port_library_list        (2): found mtab fsname pstore
0.007609 gp_port_library_list        (2): found mtab fsname bpf
[... 70 lines cut ...]
0.257260 save_settings               (2): Saving 2 setting(s) to file "/home/kingsley/.gphoto/settings"
0.258008 gp_port_info_list_lookup_path(2): Looking for path 'usb:001,017' (9 entries available)...
0.258052 gp_port_info_list_lookup_path(2): Starting regex search for 'usb:001,017'...
0.258089 gp_port_info_list_lookup_path(2): Trying '^disk:'...
0.258209 gp_port_info_list_lookup_path(2): re_match failed (-1)
0.258245 gp_port_info_list_lookup_path(2): Trying '^ptpip:'...
0.258328 gp_port_info_list_lookup_path(2): re_match failed (-1)
0.258364 gp_port_info_list_lookup_path(2): Trying '^ip:'...
0.258442 gp_port_info_list_lookup_path(2): re_match failed (-1)
0.258477 gp_port_info_list_lookup_path(2): Trying '^serial:'...
0.258561 gp_port_info_list_lookup_path(2): re_match failed (-1)
0.258597 gp_port_info_list_lookup_path(2): Trying '^usb:'...
0.258721 gp_port_info_list_get_info  (2): Getting info of entry 4 (10 available)...
0.258764 gp_camera_set_port_info     (2): Setting port info for port 'Generic Port' at 'usb:001,017'...
0.269387 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.269526 gp_port_set_settings        (2): Setting settings...
0.269571 gp_libusb1_update           (2): (old int=0, conf=-1, alt=-1) port usb:001,017, (new int=0, conf=-1, alt=-1) port usb:001,017
0.269607 gp_libusb1_update           (2): lowlevel libusb1 port not yet opened, no need for libusb changes
0.269652 gp_setting_set              (2): Setting key 'port' to value 'usb:001,017' (gphoto2)
0.269696 save_settings               (2): Saving 2 setting(s) to file "/home/kingsley/.gphoto/settings"
0.270489 gp_camera_init              (2): Initializing camera...
0.270610 gp_port_set_error [gphoto2-port.c:1190](0): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-52: 'Could not find the requested device on the USB port') ***       

*************************************

axxel, 2024-09-23:
My guess would be, you set your exposure mode dial to a mode where you don't have that property, like 'M' or 'B'. Please try with e.g. 'P' and see what happens.

sinardet, 2024-09-23:
Thanks for your answer. As you said of course I know that command :gphoto2 --get-config exposurecompensation doesn't work in Manual mode but is there anychance to get it works because I need absolutely to keep that mode?

axxel, 2024-09-23:
> is there anychance to get it works because I need absolutely to keep that mode?

I don't understand this sentence. The camera does not support setting exposure compensation in Manual mode. Neither via the camera controls, nor via USB remote.

Can you confirm that setting it works if you set the mode dial to 'P'?

sinardet, 2024-09-24:
yes it works

axxel, 2024-09-24:
> yes it works

That is good. So your problem is solved, right?

sinardet, 2024-09-25:
Not really but I haven't no other choice to use Manual mode for that I need to do, anyway thanks for your help.

axxel, 2024-09-25:
Well, exposure compensation tells the automatic exposure algorithm of the camera that it should offset its optimal value to make the image darker or brighter. In Manual mode, there is no automatic exposure algorithm at play. Please close this issue.

closed 2024-09-25 by sinardet