gphoto/libgphoto2 #1018: First capture takes too long. Further captures are fine.

sygolene, 2024-09-16:
**Describe the bug**
Hello, I'd like to submit the following issue, I didn't find a recent ressembling issue after a search.

The first capture always takes much longer than subsequent ones.
first capture : ~10s (20+ in debug)
next : < 0.2s
Tested using the python binding python-gphoto2 & command line gphoto2.

The camera is set to manual mode and manual focus so there aren't scene related influence on the capture. When untethered, the camera captures fine with no delays on first capture.

gphoto2 --debug --debug-loglevel=debug --debug-logfile=logfile.txt --capture-image -I 1 -F 5
Time-lapse mode enabled (interval: 1s).                                        
Capturing frame #1/5...
New file is in location /store_00010001/DCIM/101OLYMP/P1011065.JPG on the camera
not sleeping (23 seconds behind schedule)
Capturing frame #2/5...
New file is in location /store_00010001/DCIM/101OLYMP/P1011066.JPG on the camera
not sleeping (0 seconds behind schedule)

Jim Easterbrook & I narrowed it down to a likely libgphoto2 issue.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB ID
camera ID has always been weird here : detecting E-M1 but in use, properly recognizes the actual camera (E-M5 mkII) and the capture

- From shell

$gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Olympus E-M1                   usb:003,003

- From debug files, proper detection

Device info:
Manufacturer: OLYMPUS
Model: E-M5MarkII
device version: 1.00
serial number: 'BHEA54750'

**libgphoto2 and gphoto2 version**
libgphoto2 2.5.31
gphoto2 2.5.28

**To Reproduce**
Steps to reproduce the behavior:
capture a single frame using gphoto2 : gphoto2 --capture-image
or
an interval shoot : gphoto2 --capture-image -I 1 -F 5
See attached debug file. 
[logfile-debug.txt](https://github.com/user-attachments/files/17011859/logfile-debug.txt)
Looking at the timecodes, we see subsequent frames speeding fine.

Thanks in advance for your consideration and help !

sygolene, 2024-09-18:
the capture time was actually growing progressively. I formatted the SD card used by the camera and it solved the problem.

closed 2024-09-18 by sygolene

msmeissn, 2024-09-18:
(yes, background is to give the correct filename, we need to read the filesystem info of the whole card. After first capture the filesystem is cached, so only the first capture is slow.)

sygolene, 2024-09-18:
I see, the file system was ever growing, making the traverse longer. Interesting. Thanks msmeissn !
Wouldn't there be an option to concede the naming to the camera, as I'm saving on the SD card.