gphoto/libgphoto2 #1019: About support for EOS R50 ?

ClemSc, 2024-09-16:
Hello,

it is kind of a bottle to the sea. I have asked on Reddit and Canon community forum for someone who would be able to check whether the Canon EOS R50 is supported by libgphoto2 (it is not on the official support list, but this list might not be complete ?).

I need to buy a few cameras from a budget, this model fits all requirements, but gphoto2 support would be necessary. Unfortunately, the combination "R50 owner" / "gphoto2" user seems really uncommon.
Perhaps someone can comment here ? The R10 is supported, does that says anything about the likelihood of the R50 to also work ?

Best.

msmeissn, 2024-09-16:
It is highly likely just supported out of the box.
For Canon EOS and others we usually do not need to have it specifically added to be supported.

SO should work.

matthewstent, 2024-09-19:
Can confirm it works!

msmeissn, 2024-09-29:
thanks for the confirmation!

closed 2024-09-29 by msmeissn

ClemSc, 2024-10-23:
@matthewstent 
> Can confirm it works!

Well, I have now the camera. And I struggle. This is my version of gphoto2 and libgphoto2.

> This version of gphoto2 is using the following software versions and options:
> gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
> libgphoto2      2.5.31         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, no EXIF
> libgphoto2_port 0.12.2         iolibs: disk ptpip serial, gcc, no ltdl, no EXIF, no USB, serial without locking

I connect the camera to the computer using usb. 
gphoto2 --auto-detect gives nothing.

I also tried with python-gphoto2. This is what I get as camera summary:
```
Summary
=======
Manufacturer: Canon.Inc
Model: Canon EOS R50
  Version: 3-1.2.0
  Serial Number: f941a752d141f0757b3489afb115e780
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description: 

Capture Formats: JPEG Unknown(b108) Unknown(b10b) Unknown(b982)
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b105), Unknown(b982), Unknown(b10a), Unknown(b10b), Unknown(b109)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS R50'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)
```

If I try to capture a picture I get this error:

> GPhoto2Error: [-110] I/O in progress

So, how did you made it work ? :)