gphoto/libgphoto2 #1028: Support for Canon EOS R5m2 | Canon EOS R5 Mark II

karlkopf, 2024-09-24:
**Name of the camera*

Canon EOS R5 Mark II

**USB IDs**
Bus 002 Device 005: ID 04a9:3314 Canon, Inc. Canon Digital Camera
<details>
  <summary>lsusb -v</summary>
  
  ```C
Bus 002 Device 005: ID 04a9:3314 Canon, Inc. Canon Digital Camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x04a9 Canon, Inc.
  idProduct          0x3314
  bcdDevice            0.02
  iManufacturer           1 Canon Inc.
  iProduct                2 Canon Digital Camera
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass         6 Imaging
      bInterfaceSubClass      1 Still Image Capture
      bInterfaceProtocol      1 Picture Transfer Protocol (PIMA 15470)
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
[... 1031 lines cut ...]
sys     0m0.122s
  ``` 

  gphoto2 --capture-tethered
```C
time env LC_ALL=C gphoto2 --debug-logfile=capture-tethered_logfile.log --debug --capture-tethered --filename=raw_current_%Y-%m-%d_%H.%M.%S_%06n.jpg --hook-script=/usr/local/sbin/hook-script
Waiting for events from camera. Press Ctrl-C to abort.
*** Error (-10: 'Timeout reading from or writing to the port') ***

^C
Cancelling...

real    0m50.971s
user    0m0.430s
sys     0m0.143s
  ``` 

At the moment it's not possible for me to capture pictures from the camera

[capture-image-and-download_logfile.log](https://github.com/user-attachments/files/17115391/capture-image-and-download_logfile.log)
[capture-preview_logfile.log](https://github.com/user-attachments/files/17115392/capture-preview_logfile.log)
[capture-tethered_logfile.log](https://github.com/user-attachments/files/17115393/capture-tethered_logfile.log)
[list-all-config.txt](https://github.com/user-attachments/files/17115394/list-all-config.txt)
[list-all-config_logfile.log](https://github.com/user-attachments/files/17115395/list-all-config_logfile.log)
[summary_logfile.log](https://github.com/user-attachments/files/17115396/summary_logfile.log)

axxel, 2024-09-26:
This is weird. It times out while waiting for the Response packet of a `EOS_GetDeviceInfoEx` call after the data has been transferred apparently successfully (I compared the output with my R8 and the output seems reasonable).

I noticed you were working with a fairly up to date code base. Can you quickly test if `gphoto2 --get-config aperture` fails with both the current HEAD revision as well as with the last release version [2.5.31](https://github.com/gphoto/libgphoto2/releases/tag/v2.5.31)?

karlkopf, 2024-09-26:
> This is weird. It times out while waiting for the Response packet of a `EOS_GetDeviceInfoEx` call after the data has been transferred apparently successfully (I compared the output with my R8 and [...]

I will do any test you want. I just have to wait to get the camera back

karlkopf, 2024-09-27:
I just testet a small set of actions on the camera with different commits and branches but I wasn't able to get a picture from the camera. I'll upload the logs to a git repository and link it here later

axxel, 2024-09-27:
Thanks for the effort. I conclude that
 1) `EOS_GetDeviceInfoEx` fails to send a PTP response packet even on an 8 month old code base as well as with the latest.
 2) my new `imageformat2` code works for the R5m2

My first guess is: this is a bug in the firmware of the camera. It just so happens that Canon released a 1.0.1 [firmware update](https://www.canonrumors.com/canon-eos-r5-mark-ii-firmware-v1-0-1-released/) yesterday. Could you maybe try updating it and then perform another `gphoto2 --get-config aperture`?

axxel, 2024-09-27:
If the firmware upgrade is not an option or does not help, please see if the following 2 lines added [exactly here](https://github.com/gphoto/libgphoto2/blob/042c4f38092bbffe95b490df93f37cd607c3c3d1/camlibs/ptp2/ptp.h#L4828) does fix it for you.
```c
if (operation == PTP_OC_CANON_EOS_GetDeviceInfoEx && !strcmp(params->deviceinfo.Model,"Canon EOS R5m2"))
    return 0;
```

msmeissn, 2024-09-27:
can i also ask to check if hardware is affected, like switch to a different usb port, use a different cable?

karlkopf, 2024-09-27:
Thanks for all your suggestions.

I have updated the camera to FW: 1.0.1 but there was no big change in the behavior

In the next step I patched the ptp.h and testet it with the different commits --> this was a little success! :-)

Now the camera informations from for example --list-all-config appears immediately and not after 3m20s
It is also possible to capture a preview, but after the capture the camera is in "busy" mode

It's still not possible for me to do --capture-and-download and --capture-tethered

I have tried three different USB cabel to be sure

If I can further assist you, I'd be happy to do so

axxel, 2024-09-27:
> I have updated the camera to FW: 1.0.1 but there was no big change in the behavior

Ahh, bummer. That would have been too easy ;).
 
> In the next step I patched the ptp.h and testet it with the different commits --> this was a little success! :-)

So far so good. The hanging, now disabled call is actually not really necessary for the devices operation.

> It's still not possible for me to do --capture-and-download and --capture-tethered

The capture command now fails with busy when sending the "Full press button" command after the library went into a 2s timeout waiting for some feedback about successful auto-focusing. Could you please set your lens / settings to manual focus, just to see if that changes things?

karlkopf, 2024-09-27:
> The capture command now fails with busy when sending the "Full press button" command after the library went into a 2s timeout waiting for some feedback about successful auto-focusing. Could you please set your lens / settings to manual focus, just to see if that changes things?

I tested it with manual focus, after a --capture-preview the camera is still in busy mode
Test with --capture-image-and-download - Now there is a shutter sound and an image is taken on the camera but gphoto is unable to retreive the image.

axxel, 2024-09-27:
Looks like we are making slow progress... ;). The log-file indicates that there is no SD-Card in the camera. Just to rule out more potential pitfalls: can you please insert an SD-Card? Then please set the mode dial to 'M', the exposure time to 1/60  execute the following command:
```sh
gphoto2 --set-config capturetarget=1 --trigger-capture --wait-event=1s
```

karlkopf, 2024-09-28:
Now the image is triggered and will be saved on the SD-Card but not downloaded by gphoto2

--get-all-files also doesn't work

msmeissn, 2024-09-30:
it looks weird that commands fail, and usually it would mean hardware or firmware issues.

so, yes, check if it works on windows, and if yes, try to get a packettrace (using USBPcap for instance)

karlkopf, 2024-09-30:
Thanks for the info

I have added a raw USB trace.

I have started the trace, opened the Canon EOS Utility, selected remote capture, took a picture, close the utility

I hope this'll help a bit

msmeissn, 2024-09-30:
One thing i see is it calls 0x9170, the GetObjectInfo64 call  ... We are not doing that so far.
calls also 913f PTP_OC_CANON_EOS_GetCameraSupport

There is 0x9087 which I do not know the name for yet.

and 0x917d whuich I also do not know.

msmeissn, 2024-09-30:
[remote-capture.txt](https://github.com/user-attachments/files/17187948/remote-capture.txt)

axxel, 2024-09-30:
I spend all afternoon manually parsing the dumped data (reminded me of the good old times, when I did this for the 5Dm2 ;)). I'm not fully done, yet. But I think I found the key to maybe get the download going after capture: a new EOS event `0xc1b8`. @msmeissn where did [this name](https://github.com/gphoto/libgphoto2/blob/56885c02c8ec9641aa85d33996272d577713a7c7/camlibs/ptp2/ptp.h#L1286) come from? Could it be that there is a typo and this was actually supposed to be `c1b8` instead of `c1a8`? Or could you get some more of those names, which would already help quit a bit in deciphering their meaning.

Not clear yet why we ran into the busy issue when the in auto-focus mode. @karlkopf was that USB trace with manual or auto-focus? Can you make a second remote-capture trace with the setting reversed?

msmeissn, 2024-09-30:
I think a name in the disassembly of Canon EOS driver kit.

karlkopf, 2024-10-01:
> Not clear yet why we ran into the busy issue when the in auto-focus mode. @karlkopf was that USB trace with manual or auto-focus? Can you make a second remote-capture trace with the setting reversed?

Yesterday it was done with manual focus, shutter time 1/60

Now I have done three traces, manual focus, auto focus and auto focus fail. When the auto focus fail you can't capture an image with the eos utility --> status busy

axxel, 2024-10-01:
@karlkopf thanks for the new trace data. I'll have a look at it later. Can you do me the favor and pull current HEAD and run these commands with manual focus and 1/60?: 
```
gphoto2 --set-config capturetarget=4 --trigger-capture --wait-event=1s
gphoto2 --set-config capturetarget=1 --trigger-capture --wait-event=1s
```
I improved the event logging to help with finding the differences between EOSUtility and gphoto2 faster. This also includes the one-liner fix, so it should do something out of the box. If you feel bored, doing the first with auto-focus would also be of interest.

karlkopf, 2024-10-01:
@axxel 
I just compiled the latest head, run a few tests and added the new debug logs for af and mf. I hope this'll help a little bit

axxel, 2024-10-01:
@karlkopf Thanks for all the new data. Unfortunately, the new event `c1b8` does not show up in any of the logs. So the EOS utility does something important before performing the remote-release commands. It might be settings the CaptureDestination to 4 (instead of 2). I thought I could achieve that with the `capturetarget=4` but I was mistaken.

Random idea: you are not living in Berlin by any chance? That would maybe open the possibility for us meeting in person and having your camera being connected to my PC and thereby speeding up this trial-and-error processes substantially.

axxel, 2024-10-01:
Here is my current "manual" parsing of the original USB message trace:
```
0110 - GetDeviceInfo
0191 - GetStorageIDs
0291 - GetStorageInfo
1091 - SetDevicePropValueEx
1491 - SetRemoteMode
1591 - SetEventMode
1791 - TransferComplete
1a91 - PCHDDCapacity
2791 - RequestDevicePropValue
2891 - RemoteReleaseOn
2991 - RemoteReleaseOff
3d91 - SetRequestOLCInfoGroup

new/unused

3f91 - GetCameraSupport
7091 - GetObjectInfo64
7291 - GetPartialObject64
7d91 - ???
8790 - ???
8991 - GetPictureStylePCFlavorParam

===============================================

actual order of appearance with arguments:
 *'#' means a command/param that we don't use, yet
 * GetEvent (9116) has been ommitted most of the time to make it more readable

1491 01000000 					- SetRemoteMode 1
1591 01000000					- SetEventMode 1
7091 000020a3					# GetObjectInfo64
7091 000020a3					# GetObjectInfo64
7091 000020a3					# GetObjectInfo64
0291 00000200					- GetStorageInfo
3d91 ff0f0000					- SetRequestOLCInfoGroup
3f91 00000001 10010000 00008431 		# GetCameraSupport
0191 						- GetStorageIDs
0291 00000100 					- GetStorageInfo
0291 00000200 					- GetStorageInfo
3f91 00000001 10020000 09008431 		# GetCameraSupport
0110 						- GetDeviceInfo
8790 						# ??? - this returns nothing
7d91 01000000					# ??? - this retrieves over 50kB of 'random' data

1491 02000000 					# SetRemoteMode 2
0110 						- GetDeviceInfo
2791 afd10000					- RequestDevicePropValue ???
0110 						- GetDeviceInfo
2791 a6d10000					- RequestDevicePropValue ???
2791 69d10000					- RequestDevicePropValue ErrorHistory
2791 6ad10000					- RequestDevicePropValue LensExchangeHistory
2791 6bd10000					- RequestDevicePropValue StroboExchangeHistory
2791 72d20000					- RequestDevicePropValue ???
0291 00000100					- GetStorageInfo
0291 00000200					- GetStorageInfo
2791 d0d10000					- RequestDevicePropValue Artist
2791 d1d10000					- RequestDevicePropValue Copyright

1491 01000000 					- SetRemoteMode 1
1091 						# SetDevicePropValueEx (with no parameters!?!)
1091 0c000000 1cd10000 04000000 		- SetDevicePropValueEx CaptureDestination -> 4 (PC?)

1a91 7f86fb0b 00100000 01000000 		- PCHDDCapacity
8991 01000000					- GetPictureStylePCFlavorParam
8991 02000000					- GetPictureStylePCFlavorParam
8991 03000000					- GetPictureStylePCFlavorParam
2791 7fd10000					- RequestDevicePropValue FlavorLUTParams

2891 01000000 01000000				- RemoteReleaseOn (1, 1)
1691						- GetEvent (see A)
2991 01000000					- RemoteReleaseOff (1)

# capture starts here
2891 01000000 01000000				# RemoteReleaseOn (1, 1) # use use (1, 0)
1691						- GetEvent (see B)
1a91 e1feffff 00100000 00000000	- PCHDDCapacity
0291 00000100					- GetStorageInfo
0291 00000200					- GetStorageInfo
2891 02000000 01000000				# RemoteReleaseOn (2, 1) # we use (2, 0) @33.66
1691 						# GetEvent (see C) @ 33.70
2991 02000000					- RemoteReleaseOff (2)   @ 33.95s
1691 						- GetEvent (see D1) @ 34.14
1691 						- GetEvent (see D2) @ 34.36 contains c1b8 80880714
7291 80880714 00000000 00f00f00 00000000	# GetPartialObject64 @ 34.37
7291 80880714 00f00f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00e01f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00d02f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00c03f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00b04f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00a05f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00906f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00807f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00708f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 00609f00 00f00f00 00000000	# GetPartialObject64
7291 80880714 0050af00 00f00f00 00000000	# GetPartialObject64
2991 01000000					- RemoteReleaseOff (1)
7291 80880714 0040bf00 05880c00 00000000	# GetPartialObject64
1791 80880714					# TransferComplete
1a91 b0fcffff 00100000 00000000 		- PCHDDCapacity

2891 01000000 01000000				- RemoteReleaseOn (1, 1)
2991 01000000					- RemoteReleaseOf (1)
1a91 d4ffffff 00100000 00000000 		- PCHDDCapacity

Event C:
888	33.703424	URB_BULK in	
520000000200169197000000
0c000000 8bc10000 01000000 		- CameraStatusChanged 1
10000000 89c10000 1bd10000 789f0000	- PropValueChanged (EOS_AvailableShots)
10000000 89c10000 c3d10000 00000000	- PropValueChanged (EOS_OneShotRawOn)
12000000 a5c10000 0e000000010000000400	- OLCInfoChanged
08000000 00000000

Event D1:
898	34.142530	URB_BULK in
700000000200169199000000
10000000 89c10000 b7d10000 00000000		- PropValueChanged (EOS_ExposureSimMode)
10000000 89c10000 c3d10000 00000000 		- PropValueChanged (EOS_OneShotRawOn)
12000000 a5c10000 0e000000010000000200		- OLCInfoChanged
15000000 a5c10000 11000000000200000101040000	- OLCInfoChanged
15000000 a5c10000 11000000000200000101080000	- OLCInfoChanged
08000000 00000000

Event D2:
904	34.360451	URB_BULK in
38000000020016919a000000 
24000000 b8c10000 80880714 01380000 00000000 05c8cb00 00000000 00880714 01000000 # new event
08000000 00000000
```

msmeissn, 2024-10-02:
Disassembly brings new defines:

+#define PTP_EC_CANON_EOS_ObjectAddedEx64LFN            0xc1b6
+#define PTP_EC_CANON_EOS_ObjectInfoChangedEx64LFN      0xc1b7
+#define PTP_EC_CANON_EOS_RequestObjectTransfer64LFN    0xc1b8

msmeissn, 2024-10-02:
more opcodes added, the two we see are:
+#define PTP_OC_CANON_GetDebugLog               0x9087
+#define PTP_OC_CANON_EOS_GetDebugLogEx 0x917D

zaps166, 2024-12-02:
Skimmed this thread only, so don't know if it's already implemented. We have a few new events (defines already added). These events has new data structure (but similar). The difference is in file name field.

What I checked:

- `0xc1b6` event - the file name field is moved to the bottom (below the date field) preceded by by 1-byte (string length). String is in UTF-16.
- `0xc1b8` event - the the file name field is missing, the rest is the same.

tangbl93, 2024-12-16:
event from R5m2,  only tested in wifi mode

- 0xC1B6: same as PTP_EC_ObjectAdded
- 0xC1B7: same as PTP_EC_CANON_EOS_ObjectInfoChangedEx

blaineam, 2024-12-31:
> Now the image is triggered and will be saved on the SD-Card but not downloaded by gphoto2
> 
> --get-all-files also doesn't work

I am able to get images using that flag on Canon EOS R5 Mark II Camera Firmware version 1.0.2 however --capture-and-download seems to not retrieve the files when used with --HEAD version of libgphoto2.

```➜ gphoto2 --set-config capturetarget=1 --trigger-capture --wait-event-and-download=1s --get-all-files
Waiting for 1 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property d1c3 changed, "oneshotrawon" to "0"
UNKNOWN Button 2
UNKNOWN PTP Property d103 changed, "iso" to "1600"
UNKNOWN Focus Mask 0101002000
UNKNOWN PTP Property d103 changed, "iso" to "1600"
UNKNOWN PTP Property d1b7 changed
UNKNOWN PTP Property d103 changed, "iso" to "1600"
UNKNOWN Camera Status 1
UNKNOWN StorageInfoChanged 0x00010001
UNKNOWN StorageInfoChanged 0x00020001
UNKNOWN PTP Property d11b changed, "availableshots" to "10444"
UNKNOWN PTP Property d1c3 changed, "oneshotrawon" to "0"
UNKNOWN Button 4
UNKNOWN OLCInfo event 0x0010, 4 bytes: 04 02 28 cd 
UNKNOWN OLCInfo event 0x0010, 4 bytes: 04 02 28 cc 
UNKNOWN PTP Property d1c3 changed, "oneshotrawon" to "0"
UNKNOWN Button 2
UNKNOWN PTP Property d1c3 changed, "oneshotrawon" to "0"
UNKNOWN Button 3
UNKNOWN PTP Property d102 changed, "shutterspeed" to "1/125"
UNKNOWN PTP Property d103 changed, "iso" to "1600"
UNKNOWN PTP Property d1b7 changed
CAPTURECOMPLETE
UNKNOWN OLCInfo event 0x0010, 4 bytes: 04 01 28 cc 

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002

*** Error ***              
error 0x2002
Saving file as 259A7716.CR3

*** Error ***              
error 0x2002
Saving file as 259A7716.HIF

*** Error ***              
error 0x2002
Saving file as 259A7717.CR3

*** Error ***              
error 0x2002
Saving file as 259A7717.HIF

*** Error ***              
error 0x2002
Saving file as 259A7718.CR3

*** Error ***              
error 0x2002
Saving file as 259A7718.JPG

*** Error ***              
error 0x2002
Saving file as 259A7719.CR3

*** Error ***              
error 0x2002
Saving file as 259A7719.JPG

*** Error ***              
error 0x2002
^C
Cancelling...

*** Error ***              
PTP Cancel Request
Operation cancelled.```

I would like to help collect any information you can use to get a new version of libgphoto2 out that will support the Canon R5 Mark II.

Overall it is working better with the current HEAD of this repo but seems like some things are not quite working yet like capture and download doesn't find anything but does take the picture.

runrunh, 2025-05-05:
I updated firmware to 1.0.3, but I have the same issue on capturing and downloading the images. Is there any plan of a newer release that could solve this issue?

axxel, 2025-10-20:
To the R5m2 users reading this: any suggestions regarding #1171 ?

axxel, 2025-11-25:
> I don't know what those generic errors are but it shows those a lot in anything I run on latest stable version of gphoto2 installed with brew.

Those errors are gone with the above changeset. The error condition itself is still there (`ptp_canon_eos_getobjectinfoex` still fails for an unknown reason) but it is not reported anymore to the console. It is also non-critical, since libgphoto2 falls back to a generic (slower) solution that does work on the R5m2.

blaineam, 2025-12-12:
> Those errors are gone with the above changeset. The error condition itself is still there (`ptp_canon_eos_getobjectinfoex` still fails for an unknown reason) but it is not reported anymore to the [...]

How did you get the R5M2 to download the image to the computer using --capture-image or --capture-image-and-download. mine waits a while after finishing capturing the image (audible shutter sounds) and then shows this output: 
```
➜ gphoto2 --capture-image-and-download
ERROR: Could not capture image.                                                
ERROR: Could not capture.

~ took 1m 33s 

➜ sudo gphoto2 --debug --capture-image
ERROR: Could not capture image.                                                
ERROR: Could not capture.

~ took 1m 33s 
```

This was both for R5M2 Firmware 1.1.1 as well as 1.2.0

I have yet to make it work properly.

I use a 2TB SD and/or a 4TB CFe card. I have tried previously with a 2TB CFe card too. whatever I do it fails to read that the capture was successful. 

I also shoot jpg + raw files and choosing CFe or SD in camera does not make a difference with this issue. 

The 
```
*** Error ***              
error 0x2002
``` 
in the logs are gone at least.

dov, 2026-10-06:
**Canon EOS R6 Mark III (04a9:3323) still fails and hangs on libgphoto2 2.5.34, and the camera hangs afterwards**

Environment: Fedora 44, libgphoto2 2.5.34 built from the `v2.5.34` tag, gphoto2 2.5.28, camera on USB directly connected to the host. Capture target: Internal RAM.

Detection and read-only queries work (`--auto-detect` shows "Canon EOS R6 Mark III", `--get-config batterylevel` returns 74%). But `gphoto2 --capture-image-and-download` fails:

```
*** Error ***
Canon EOS Half-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
*** Error (-110: 'I/O in progress') ***
```

Debug log (attached): `EOS_RemoteReleaseOn (0x1,0x0)` is sent at 0.124 s. The camera answers `0x2019` only at 5.120 s, after the 5000 ms port timeout. No `EOS_RemoteReleaseOff` is sent after the failure, only `EOS_AfCancel`. The exit sequence ends with `SetRemoteMode(1)` before `Close session`.

In an earlier run with the lens out of focus, the half-press succeeded and the log showed `Auto-Focus failed`, followed by `Full-Press failed (0x2019: PTP Device Busy)`.

After these failures the camera became unresponsive and needed a power cycle (`PTP Timeout`, or it vanished from `lsusb`). It happened again after a session that only ran `--auto-detect` and `--get-config`, with no capture.

If there is any other info that I can provide that may help solve this bug, please let me know.

[my-logfile2.txt](https://github.com/user-attachments/files/33125554/my-logfile2.txt)