gphoto/libgphoto2 #1032: How to cross-compile libgphoto2 (to aarch64 on Ubuntu/Debian)?

axxel, 2024-09-27:
I'm trying to cross-compile to aarch64. I installed `crossbuild-essential-arm64` and now have a working compiler and toolchain, meaning I can compile a hello-world program and run it happily.

Next step was
```sh
configure --build x86_64-pc-linux-gnu --host aarch64-linux-gnu --enable-mpers=check
    --disable-nls --without-jpeg --with-libxml-2.0=no --with-libexif=no --with-libcurl=no --with-gdlib=no
    --disable-serial --with-camlibs=ptp2
    --disable-shared --enable-static  CFLAGS="-g"
```
trying to make it as easy as possible by removing all unnecessary dependencies. configure fails with the following error:
```sh
configure: error: 
libgphoto2 requires libltdl (a part of libtool)
```
Why is `libltdl` still required if I `--disable-shared`?

I'd appreciate any hints towards either:
 * get configure find the missing libltdl
 * compile libgphoto2/libgphoto_port with statically linked ptp2 camlib and libusb driver, to remove the libltdl dependency

Maybe @ndim or @hfiguiere have some relevant insights?

ndim, 2024-09-27:
Whether you build libgphoto2 and libgphoto2_port as a static or as a dynamic library is not the point here. They both use libltdl to dybamically load camlibs (camera drivers) or iolibs (port drivers), respectively.

axxel, 2024-09-27:
Thanks for the clarification. Maybe the real question is: why? What is the use case for having a static build that still loads dlls at runtime?

Is there maybe some "hack" that could be applied to the build system to link those symbols statically into the library?

ndim, 2024-09-27:
The variables to set for building with your custom libltdl:

```
$ ./configure --help | grep -i ltdl
  LTDLINCL    CFLAGS for compiling with libltdl
  LIBLTDL     LIBS to add for linking against libltdl
$ _
```