gphoto/libgphoto2 #1043: Please support NIKON Z6iii

2024-jl, 2024-10-17:
Fill in fields below. If you don't know how, leave it free.

NIKON Z6 iii

Use the Marketing name.

**USB IDs**
e.g. by running:   lsusb 

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

2024-jl, 2024-10-17:
Please support NIKON Z6 iii very important for astrophotography

msmeissn, 2024-10-19:
usually we support it in generic fashion already.

if you oown it, can you report the above data, like USB ids so we can add it for named detection?

rigodlc, 2024-10-31:
Nikon Z6iii

Use the Marketing name.

**USB IDs**
`Bus 001 Device 006: ID 04b0:0454 Nikon Corp. NIKON DSC Z6_3`

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here
[summary.txt](https://github.com/user-attachments/files/17581304/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt
[list-all-config.txt](https://github.com/user-attachments/files/17581311/list-all-config.txt)

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
  gphoto2 --capture-preview

Both of these commands work fine

msmeissn, 2024-11-11:
the camera is listed, thanks for the report!

closed 2024-11-11 by msmeissn