gphoto/libgphoto2 #1049: Support for Olympus PEN E-PL2

tjhietala, 2024-10-25:
**Name of the camera**

Olympus PEN E-PL2

**USB IDs**
ID 07b4:0113 Olympus Optical Co., Ltd Mju 500 / Stylus Digital Camera (PTP)

**camera summary output**
[summary.txt](https://github.com/user-attachments/files/17525157/summary.txt)

**camera configuration output**
[list-all-config.txt](https://github.com/user-attachments/files/17525170/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
> *** Error ***              
> Sorry, your camera does not support generic capture
> ERROR: Could not capture image.
> ERROR: Could not capture.
> *** Error (-6: 'Unsupported operation') ***    

  gphoto2 --capture-preview

> *** Error (-6: 'Unsupported operation') ***

tjhietala, 2024-11-05:
The camera does not appear to have remote capture support over USB.

msmeissn, 2024-11-10:
yes, this series is currently not known to be able to remote controll4ed capture