gphoto/libgphoto2 #1050: Support for Panasonic Lumix DMC-GX7

tjhietala, 2024-10-25:
**Name of the camera**

Panasonic Lumix DMC-GX7

**USB IDs**
ID 04da:2374 Panasonic (Matsushita) Lumix Camera (PTP mode)
ID 04da:2372 Panasonic (Matsushita) Lumix Camera (Storage mode)

The information below is from PTP mode.

**camera summary output**
  gphoto2 --summary > summary.txt
[summary.txt](https://github.com/user-attachments/files/17526333/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt
[list-all-config.txt](https://github.com/user-attachments/files/17526335/list-all-config.txt)

**test capture**

  gphoto2 --capture-image-and-download

> *** Error ***              
> Sorry, your camera does not support generic capture
> ERROR: Could not capture image.
> ERROR: Could not capture.
> *** Error (-6: 'Unsupported operation') ***

   gphoto2 --capture-preview

> *** Error (-6: 'Unsupported operation') ***

msmeissn, 2024-10-31:
The camera currently does not report capture abilities over PTP.

A large set of Panasonic / Lumix does not support remote control over USB.

If there is remote capture support under Windows or MacOS it would be good to know for us.

tjhietala, 2024-11-05:
The camera does not appear to have remote capture support over USB.