gphoto/libgphoto2 #1056: [QUESTION] Nikon S9100 - Continues HDMI live feed?

Maxsafer, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
No. I is not. And I'm sorry I asked in the request feature section.

**Describe the solution you'd like**
I really just want to ask you guys if its possible, since you have more experience and I have seen you guys know more than me about the Nikon S9100.

**Describe alternatives you've considered**
I have thought about trying to modify the firmware, though I don't know if I'm digging my own grave.

**Additional context**
I want to use this cam as a web cam.

msmeissn, 2024-11-10:
hmm, question is when should the HDMI feed continue? When the camera is attached to a computer and downloading files?

Maxsafer, 2026-11-03:
> 2\. hmm, question is when should the HDMI feed continue? When the camera is attached to a computer and downloading files?

Sorry, typo in the title: I meant continuous, i.e. a live view feed I can use as a webcam. Not related to downloading. HDMI and USB can’t be connected at the same time on this camera (per Nikon’s manual)