gphoto/libgphoto2 #1058: ony DSLR ILCE-6400 crash

knro, 2024-11-12:
**Describe the bug**

Crash when camera is waiting for event when using INDI driver. Sometimes the camera can load 1 or 2 images, only to crash next.

```
[New Thread 0x7f9e20ed00 (LWP 401898)]
[Thread 0x7f9e20ed00 (LWP 401898) exited]
[New Thread 0x7f9e20ed00 (LWP 402917)]
[Thread 0x7f9e20ed00 (LWP 402917) exited]
Thread 2.1 "indi_sony_ccd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fa9ee8b80 (LWP 396659)]
__GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3362
#0  __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3362
#1  0x0000007fa6873a94 in ?? () from /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.31.1/ptp2.so
#2  0x0000007fa688f77c in ?? () from /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.31.1/ptp2.so
#3  0x0000007fa9b2d888 in gp_camera_wait_for_event () from /lib/aarch64-linux-gnu/libgphoto2.so.6
#4  0x0000005589b0f6a4 in ?? ()
#5  0x0000005589b0fba0 in ?? ()
#6  0x0000005589afa190 in ?? ()
#7  0x0000007fa9f9b47c in INDI::CCD::ISNewNumber(char const*, char const*, double*, char**, int) () from /lib/aarch64-linux-gnu/libindidriver.so.2
#8  0x0000007fa9f92e24 in ISNewNumber () from /lib/aarch64-linux-gnu/libindidriver.so.2
#9  0x0000007fa9f8b0b0 in dispatch () from /lib/aarch64-linux-gnu/libindidriver.so.2
#10 0x0000007fa9f8d184 in ?? () from /lib/aarch64-linux-gnu/libindidriver.so.2
#11 0x0000007faa038a04 in ?? () from /lib/aarch64-linux-gnu/libindidriver.so.2
#12 0x0000007faa038be0 in eventLoop () from /lib/aarch64-linux-gnu/libindidriver.so.2
#13 0x0000007fa9f60ef8 in main () from /lib/aarch64-linux-gnu/libindidriver.so.2
#14 0x0000007fa9577740 in __libc_start_call_main (main=main@entry=0x7fa9f60de0 <main>, argc=argc@entry=1, argv=argv@entry=0x7fe1599db8) at ../sysdeps/nptl/libc_start_call_main.h:58
#15 0x0000007fa9577818 in __libc_start_main_impl (main=0x7fa9f60de0 <main>, argc=1, argv=0x7fe1599db8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:360
#16 0x0000005589af9f70 in ?? ()
```

**Name the camera**
Sony DSLR ILCE-6400 (PC Control)

**libgphoto2 and gphoto2 version**
2.5.31.1

**To Reproduce**
Steps to reproduce the behavior:

```
gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download
New file is in location /capt0000.arw on the camera                            
Saving file as capt0000.arw
Deleting file /capt0000.arw on the camera
New file is in location /capt0001.arw on the camera                            
Saving file as capt0001.arw
Deleting file /capt0001.arw on the camera
Segmentation fault
```

When I ran under gdb, I got this:

```

Thread 1 "gphoto2" received signal SIGSEGV, Segmentation fault.
__strlen_generic () at ../sysdeps/aarch64/multiarch/../strlen.S:53
53	../sysdeps/aarch64/multiarch/../strlen.S: No such file or directory.
(gdb) bt
#0  __strlen_generic () at ../sysdeps/aarch64/multiarch/../strlen.S:53
#1  0x0000007ff7b06724 in __vfprintf_internal (s=s@entry=0x7fffffa8f8, format=format@entry=0x7ff669acf8 "ObjectInfo for '%s':", ap=..., mode_flags=mode_flags@entry=2) at ./stdio-common/vfprintf-process-arg.c:397
#2  0x0000007ff7b23cac in __vsnprintf_internal (string=0x7fffffa9e8 "ObjectInfo for '", maxlen=<optimized out>, format=0x7ff669acf8 "ObjectInfo for '%s':", args=..., mode_flags=2) at ./libio/vsnprintf.c:114
#3  0x0000007ff7ba5c54 in ___vsnprintf_chk (s=<optimized out>, maxlen=<optimized out>, flag=<optimized out>, slen=<optimized out>, format=<optimized out>, ap=<error reading variable: Cannot access memory at address 0x73>)
    at ./debug/vsnprintf_chk.c:34
#4  0x0000007ff7ed3a48 in gpi_vsnprintf () from /lib/aarch64-linux-gnu/libgphoto2_port.so.12
#5  0x0000007ff7ed3efc in gp_logv () from /lib/aarch64-linux-gnu/libgphoto2_port.so.12
#6  0x0000007ff7ed4050 in gp_log () from /lib/aarch64-linux-gnu/libgphoto2_port.so.12
#7  0x0000007ff663a3b8 in ?? () from /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.31.1/ptp2.so
#8  0x0000007ff664eefc in ?? () from /usr/lib/aarch64-linux-gnu/libgphoto2/2.5.31.1/ptp2.so
#9  0x0000007ff7f0d888 in gp_camera_wait_for_event () from /lib/aarch64-linux-gnu/libgphoto2.so.6
#10 0x00000055555650d8 in ?? ()
#11 0x0000005555565580 in ?? ()
#12 0x0000005555566540 in ?? ()
#13 0x0000007ff7c63058 in ?? () from /lib/aarch64-linux-gnu/libpopt.so.0
#14 0x0000007ff7c630e8 in ?? () from /lib/aarch64-linux-gnu/libpopt.so.0
#15 0x0000007ff7c64828 in poptGetNextOpt () from /lib/aarch64-linux-gnu/libpopt.so.0
#16 0x000000555555842c in ?? ()
#17 0x0000007ff7ad7740 in __libc_start_call_main (main=main@entry=0x5555556ac0, argc=argc@entry=4, argv=argv@entry=0x7ffffff068) at ../sysdeps/nptl/libc_start_call_main.h:58
#18 0x0000007ff7ad7818 in __libc_start_main_impl (main=0x5555556ac0, argc=4, argv=0x7ffffff068, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:360
#19 0x0000005555559ef0 in ?? ()
(gdb) 
```

The generated gphoto.txt was 211MB in size and 45MB compressed! I will see if I can upload somewhere and post a link.

knro, 2024-11-12:
Log file here: https://drive.google.com/file/d/11z8C02WLzfF6lLyN6ApINma6Z1t0YkIG/view?usp=sharing

msmeissn, 2024-11-12:
relevant part:

3.549678 camera_wait_for_event       (2): DEBUG== 0xd215 after capture = 65531
3.549682 camera_wait_for_event       (2): SONY ObjectInMemory count change seen, retrieving file
3.549693 ptp_usb_sendreq             (2): Sending PTP_OC 0x1008 (Get object info) (0xffffc001) request...
3.549699 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
3.549776 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 08 10-13 00 00 00 01 c0 ff ff  ................

3.549795 ptp_usb_getdata             (2): Reading PTP_OC 0x1008 (Get object info) data...
3.549798 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
3.550581 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 02 00 08 10-13 00 00 00              ............

3.550614 ptp_usb_getresp             (2): Reading PTP_OC 0x1008 (Get object info) response...
3.550618 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
3.550838 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-13 00 00 00              ....... ....

getobjectinfo returns a very short data blob

did this work with 2.5.31?

msmeissn, 2024-11-12:
likely also failed with libgphoto 2.5.31.

the code does not expect an invalid objectinfo here though

msmeissn, 2024-11-12:
i tried fixing the crash first. but it still likely is broken