gphoto/libgphoto2 #1060: regression in imageformat for Canon EOS M10

msmeissn, 2024-11-21:
**Describe the bug**

--get-config imageformat

returns

Label: Image Format
Readonly: 0
Type: RADIO
Current: Large Normal JPEG
Choice: 0 Large Normal JPEG
END

in 2.5.31

In current GIT:
Label: Image Format                                                            
Readonly: 0
Type: RADIO
Current: (null)
END

**Name the camera**

Canon EOS M10

**libgphoto2 and gphoto2 version**

2.5.31.1 current GIT vs 2.5.31

**To Reproduce**
Steps to reproduce the behavior:

gphoto2 --get-config imageformat

msmeissn, 2024-11-21:
[eos.log](https://github.com/user-attachments/files/17846400/eos.log)  2.5.31 logfile
[eos-new.log](https://github.com/user-attachments/files/17846401/eos-new.log) 2.5.31.1 logfile

msmeissn, 2024-11-21:
but perhaps axxel has a better fix

msmeissn, 2024-11-22:
Yes, it does not return an enumeration, but a value. 

Might depend on the camera mode, but the camera is currently in Manual Mode

axxel, 2024-11-22:
Apart from the code duplication in your fix, I don't see a better approach then. I'll post a cosmetic fix for that at some point in time.