gphoto/libgphoto2 #1061: regression in sdram capture on EOS 2000d

msmeissn, 2024-11-21:
**Describe the bug**

A clear and concise description of what the bug is.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

msmeissn, 2024-11-21:
bisects to
6a1a3fa672e25c522de13ad24ea6a39546affaa0 is the first bad commit

commit 6a1a3fa672e25c522de13ad24ea6a39546affaa0
Author: axxel <awagger@gmail.com>
Date:   Tue Oct 22 11:59:30 2024 +0200

    ptp2: cleanup of find_child() and [folder|file]_list_func() functions
    
    Those 3 functions were calling ptp_list_folder to update the object cache
    and then iterating over the object cache to find the children of some
    specific handle, thereby calling ptp_object_want() which could potentially
    remove objects from the list they were iterating over. This was explicitly
    marked as "DANGEROUS", which it was.
    
    This patch introduces a return parameter to ptp_list_folder, to let the
    caller know, which handles it found. This does two things: a) it is a
    potential performance improvement and b) the caller can iterate over a
    fixed list of handles, instead of a potentially changing list of objects.
    This also did away with the 'retry' logic that tried to deal with this
    situation.
    
    I further removed the special fetching of the root folder entries from
    the camera_init function and other places and I removed the code
    duplication between file_list_func and folder_list_func.

msmeissn, 2024-11-21:
hmm, does not happen al the time, so bisect is mood

msmeissn, 2024-11-22:
another bisect also kind of found commit 6a1a3fa672e25c522de13ad24ea6a39546affaa0

axxel, 2024-12-21:
Sorry but I completely missed this issue and only found it by chance. Please make sure to mention my handle in case problems with my code show up. Since I don't have the mentioned hardware. Could you maybe provide two logs: one showing the failure and one from the commit just before the suspecting one?

Also what does "kind of found" mean?

closed 2025-07-06 by msmeissn