gphoto/libgphoto2 #1063: Nikon 1 J3 exposure time not correct

HaraldStauss, 2024-11-22:
**Describe the bug**
A clear and concise description of what the bug is.

Yesterday I posted this bug in the gphoto2 github. But I think it should be posted here. Sorry for posting it twice.
gphoto2 correctly detects the Nikon 1 J3 camera. It also sets the exposure time correctly using:
gphoto2 --set-config Shutterspeed=15
This sets the exposure time to 15 seconds (confirmed on camera display)
After that:
gphoto2 --capture-image
takes an image but only with an exposure time of less than 1 second.

I looked though the code of the camlibs/ptp2/library.c file and noticed that there have been some changes made related to the J1/V1/S1 cameras. I wonder if similar changes are needed for the J3. Thank you for looking into this.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

Nikon J3 usb:001,010

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2 2.5.28 gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2 2.5.30 standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1 iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

connect camera to USB port
gphoto2 --set-config Shutterspeed=15
gphoto2 --capture-image

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.
[Nikon_J3_Debug.txt](https://github.com/user-attachments/files/17877162/Nikon_J3_Debug.txt)

msmeissn, 2024-11-23:
dup of https://github.com/gphoto/gphoto2/issues/661

msmeissn, 2024-11-23:
there is also a shutterspeed2 option perhaps, can you check that?

can you also attach gphoto2 --summary   output?

HaraldStauss, 2024-11-23:
First of all, thank you so much for looking into this problem and also for maintaining this very important library!!!
gphoto2 --set-config shutterspeed2=5
works and successfully changes the shutter speed.

I made the following changes in camlibs/ptp2/library.c
		  /* V1 and J1 are a less reliable then the newer 1 versions, no changecamera mode, no getevent, no initiatecapturerecinsdram */
		  if (	!strcmp(params->deviceinfo.Model,"V1") ||
			!strcmp(params->deviceinfo.Model,"S1") ||
			!strcmp(params->deviceinfo.Model,"J1") ||
			!strcmp(params->deviceinfo.Model,"J3")
			) {
		    /* on V1 and J1 even the 90c7 getevents does not work */
		    /* V1: see https://github.com/gphoto/libgphoto2/issues/569 */
		    /* J1: see https://github.com/gphoto/libgphoto2/issues/716 */
		    /* S1: see https://github.com/gphoto/libgphoto2/issues/845 */
		    for (i=0;i<di->Operations_len;i++) {
		      if (strcmp(params->deviceinfo.Model,"J3") && di->Operations[i] == PTP_OC_NIKON_GetEvent) {
			GP_LOG_D("On Nikon S1/J1/J3/V1: disable NIKON_GetEvent as its unreliable");
			di->Operations[i] = PTP_OC_GetDeviceInfo; /* overwrite */
		      }
		      if (strcmp(params->deviceinfo.Model,"J3") && di->Operations[i] == PTP_OC_NIKON_InitiateCaptureRecInSdram) {
			GP_LOG_D("On Nikon S1/J1/J3/V1: disable NIKON_InitiateCaptureRecInSdram as its unreliable");
			di->Operations[i] = PTP_OC_InitiateCapture; /* overwrite */
		      }
		      if ((!strcmp(params->deviceinfo.Model,"S1") || !strcmp(params->deviceinfo.Model,"J3")) &&
			  (di->Operations[i] == PTP_OC_NIKON_InitiateCaptureRecInMedia))
			{
			  GP_LOG_D("On Nikon S1/J1/J3/V1: disable NIKON_InitiateCaptureRecInMedia as its unreliable");
			  di->Operations[i] = PTP_OC_InitiateCapture; /* overwrite */
			}
		    }
		  } else {
		    di->Operations[di->Operations_len+0] = PTP_OC_NIKON_ChangeCameraMode;
		    di->Operations_len++;
		  }
		}

With these changes, 
gphoto2 --capture-image
seems to capture with the correct exposure time. I need to do more tests to definitely confirm that.

While gphoto2 works with these changes, KStars/Ekos did not succeed in capturing images. 
I wonder if I need to recompile INDI and KStars/Ekos after changing libgphoto2.
I am trying this out right now.

Thanks,
Harald

HaraldStauss, 2024-11-23:
I forgot to attach the output of --summary. Here it is.
Is there a way to find out which functions should or should not be overwritten.
My previous fix is a trial and error and I am not sure if it is the right way.

Thanks again,

Harald
[gphoto2_--summary.txt](https://github.com/user-attachments/files/17881844/gphoto2_--summary.txt)

HaraldStauss, 2024-11-23:
OK. I did some more testing and the attached library.c file works with the Nikon J3 camera.
I recompiled INDI and now KStars/Ekos takes pictures with the J3.
Of course, Bulb does not work. So I am limited to only 30s exposure times. But I can live with that.
Perhaps you can include the changes from the attached library.c with the original distribution of libgphoto2. 
Once more, thank you so much for all your work on this very important library. It is highly appreciated. 

Thanks,

Harald
[library.c.txt](https://github.com/user-attachments/files/17882538/library.c.txt)

HaraldStauss, 2024-11-24:
[image]

I thought I share the image I took with the camera and KStars/Ekos and of course the libgphoto2 library last night. This is a stack of 63 images each 30s exposure time. 
Now of course, I wished bulb mode would work to obtain longer exposure times.
Using gphoto2 --set-config Shutterspeed=0 I can activate blub mode. If there were OP codes to trigger and release the shutter, bulb mode could be implemented. 
Do you know if there are such OP codes to press and release the shutter?
Thanks,
Harald

HaraldStauss, 2024-11-28:
Attached is a patch file for the most recent version of the library.c file. Perhaps you can update the file accordingly to make the Nikon Series 1 J3 camera work.
Thanks,
Harald
[library.patch](https://github.com/user-attachments/files/17943609/library.patch)

msmeissn, 2024-11-28:
hmm, for bulb we have code like

--set-config bulb=1  --wait-event=1000s --set-config bulb=0 

did you try that already?
(That however does set the Bulb shutterspeed.)

HaraldStauss, 2024-11-30:
Thank you for including the patch.
I compiled it based on the current github sources and the J3 works very well.
I will play more with the bulb settings and report back if I find something out that works.

Thanks,

Harald