gphoto/libgphoto2 #1064: Sudo Required To Successfully Run gphoto2 Command on Canon 5DS R from MacOS 14.1 and Above via USB

vepicurean, 2024-11-23:
I have discovered that to consistently run the standard gphoto2 commands I've been using for years on Macs with the Canon 5DS R cameras via usb, the Macs running above MacOS 14.1 and newer need to always use "sudo gphoto2" to avoid running into the dreaded "can not claim usb".  In this case the can not claim usb is caused by a kernel driver that won't detach. The only way I can get it to detach is to turn the camera off and back on. 

[Others have been discussing this sudo issue on MacOS 14.1 and above with gphoto2 for a while](https://github.com/gphoto/gphoto2/issues/562), and I see work arounds, I am likely just missing it, but I've looked through the discussions, but I don't see if there is a permanent development within gphoto2 so that sudo is not required with gphoto2 to consistently ( remote capture and download using --wait-event-and-download="FILEADDED" repeatedly, every 7 to 15 seconds) run a gphoto2 command on a Mac running MacOS 14.1 or newer? 

In summary, to get MacOS and gphoto2 to work without sudo like it was prior to MacOS 14.1, is this developed into a recent version of gphoto2 intened to run on the latest versions of MacOS ( MacOS 15 was released back in September 2024. Seem just fine, I have not tried gphoto2 on it yet)?  

That is the question I'm asking. These commands I've been using for a while so I don't think they are wrong. The underlying MacOS hardware and operating system are just not playing nice anymore with gphoto2 ( at least challenges with the MacOS software and hardware security environment via usb). 

Use an Apple Silicon Mac with legacy camera support set to on ( I don't really see that the legacy camera support toggled on does anything to help) to replicate environment exactly - others report that the sudo requirement happens on Intel Macs as well when they went to MacOS 14.1 or newer .  The Intel Macs won't be around/supported much longer, so I don't think those Macs and gphoto2 are worth worrying about much longer. The ARM CPU/APU based Macs and gphoto2 should probably be the priority. 

I'm using several Canon 5DS R at latest firmware with Macs running ARM based CPU/APU on MacOS 14.7 ( processing power and hardware setup is not the issue. Top of the line everything. All the cameras and computer hardware works perfectly independently). 

Note it is important which physical Thunderbolt ports you use when you have multiple cameras ( especially the same model of camera) connected to the same Mac - the usb port numbers have to be very distinct for gphoto2 to tell them apart. That has been the case for a while. Nothing to do with this sudo issue though). 

Command is as follows:

gphoto2 --port usb: xxy,yyx --set-config eosautofocusdrive="1" --filename somefilename.%C --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Release Full" --wait-event-and-download="FILEADDED"

Even with the standard --capture-image-and-download on the Canon 5DS R with MacOS 14.1 or newer, I have to use sudo before the gphoto2 command to get the gphoto2 command to run and override the usb port claimed roadblock. 

I use another version of the command where I've already focused the cameras and don't need to use the --set-config eosautofocusdrive="1" - Note: Make sure the lens is set to auto. Make sure the camera itself is set to either AI servo or Oneshot focus. Don't use any other AI driven focus method. 

**Note - I was getting an error unknown during the file download of the image file to the Mac. I realized that I no longer see the "FILEADDED" message in the PTP messaging like I used to. When I switch to "CAPTURECOMPLETE" on the wait-event-and-download, the error unknown went away. Using "FILEADDED" was just faster to complete the process, so I could get faster access back to the command line prompt on the Mac. Also, you don't get the error unknown on the file download with --capture-image-and-download. I just don't like using --capture-image-and-download as it causes a focus to occur when I don't need it. Less control over when to focus and when not to focus ( unless I'm mistaken). That is why I've always like the Canon EOS and gphoto2 with Mac because of the finer control over the focusing and shutter.  The users prefer Mac over Linux ( which is not surprising). But that means there are some challenges for gphoto2 as Apple does things their way and the peripheral equipment maker vendors have to fall in line. I run into it all the time.  I've seen this issues between Apple and the peripheral equipment makes at times never get resolved, which is unfortunate. QNAP storage devices and MacOS is a case in point. 

I hope these notes help. 

gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline

libgphoto2      2.5.31         standard camlibs, clang, no ltdl, EXIF

libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1, clang, no ltdl, EXIF, USB, serial without locking

vepicurean, 2024-11-24:
Some more information on why running gphoto2 with sudo causes problems for programmatic control of the cameras ( correct me if I'm missing something here). 

I finally figured out that for some reason when using sudo or running gphoto2 commands with top level su ( to get around the below mention usb device claimed), every command executed with the right usb ID, causes the usb ID to toggle ( change) back and forth. For example, when running as su, usb:002,002 changes to usb:002,003 - but when you run the next command using usb:002,003, it toggles back to usb:002,002.  The program kept telling me usb not valid, and then I finally realized what was going on. 

This behaviour does not occur when trying to run the same commands as an admin user or standard user on MacOS, but instead you get the dreaded usb device claimed (I/O error), which in this caused by a kernel driver that won't detach from the previous command.

ndim, 2024-11-26:
This now is the canonical place which describes what must be done for gphoto2 (and any libgphoto2 app at all) to seamlessly run on OSX.

Thank you very much for researching all this and writing it up.

joaopedrotaveira, 2025-06-14:
Well and finally, I manage to use digikam to import pictures from Nikon D40 and/or Canon EOS R10.

Simply kill the ptpcamerad constantly on terminal:
```sh
$ while test 1; do killall -9 ptpcamerad; sleep 1; done
```