gphoto/libgphoto2 #1068: Sony ILCE-7M3 get iso,iris,shutterspeed cannot accurately

xujinbook, 2024-12-24:
**Describe the bug**
We are using the latest version of gphoto2 to control a Sony ILCE-7M3 camera via USB and have found that we cannot accurately obtain the correct ranges for ISO, shutter speed, and iris, as they do not match those of the camera. Additionally, after changing the focal length, the retrieved value ranges also do not match those of the camera.  we very need your help, we can pay for it.

A clear and concise description of what the bug is.

**Name the camera**
If this is camera specific, include the camera name as shown by gphoto2 --auto-detect or USB IDs

**libgphoto2 and gphoto2 version**
output of:  libgphoto2 2.5.31

**To Reproduce**
Steps to reproduce the behavior:

If its a camera misbehavior, attach debug output using  --debug --debug-logfile=debug.log   (if considered useful).

jufa, 2025-01-05:
libgphoto2 2.5.30 test: replicated with ZV-E10 mk I cannot set certain shutterspeeds via value or index. Found another config that can set shutterspeed, at least on ZVE-10:
```
/main/other/d229
Label: Shutter speed
Readonly: 0
Type: MENU
Current: 69536
Choice: 0 19660810
Choice: 1 16384010
Choice: 2 13107210
Choice: 3 9830410
Choice: 4 8519690
Choice: 5 6553610
Choice: 6 5242890
Choice: 7 3932170
Choice: 8 3276810
Choice: 9 2621450
Choice: 10 2097162
Choice: 11 1638410
Choice: 12 1310730
Choice: 13 1048586
Choice: 14 851978
Choice: 15 655370
Choice: 16 524298
Choice: 17 393226
Choice: 18 327690
Choice: 19 262154
Choice: 20 65539
Choice: 21 65540
Choice: 22 65541
Choice: 23 65542
Choice: 24 65544
Choice: 25 65546
Choice: 26 65549
Choice: 27 65551
Choice: 28 65556
Choice: 29 65561
Choice: 30 65566
Choice: 31 65576
Choice: 32 65586
Choice: 33 65596
Choice: 34 65616
Choice: 35 65636
Choice: 36 65661
Choice: 37 65696
Choice: 38 65736
Choice: 39 65786
Choice: 40 65856
Choice: 41 65936
Choice: 42 66036
Choice: 43 66176
Choice: 44 66336
Choice: 45 66536
Choice: 46 66786
Choice: 47 67136
Choice: 48 67536
Choice: 49 68036
Choice: 50 68736
Choice: 51 69536
END
```

`gphoto2 --set-config-index "d229"=1`

xujinbook, 2025-01-08:
Thanks, I can get correct by the way