gphoto/libgphoto2 #1069: Directory becomes "empty" after 1st access resulting in "PTP No Device (-52)" error

Sija, 2024-12-27:
**Describe the bug**

When accessing a directory with device unplugged (via `gp_camera_folder_list_folders`), I'm getting `PTP No Device (-52)` error on the 1st try, but then on subsequent calls it returns successfully with an empty directory structure.

**Name the camera**
I've seen it happen on many cameras (PTP Generic, Canon)

**libgphoto2 and gphoto2 version**
libgphoto2 2.5.31

msmeissn, 2025-01-02:
if the camera is not present usually the api would not work anymore, it works only with the camera avaiable ...

not sure i fully understand your usecase

Sija, 2025-01-03:
Regardless of the use case, I'd expect consistency - i.e. returning the same error on every call.

pppanghu77, 2025-01-07:
How did you manage to call gp_camera_folder_list_folders without initializing the camera first? It seems like gp_camera_init would fail if no camera is plugged in, right?

Sija, 2025-01-08:
@pppanghu77 Camera is already initialized at that point.

pppanghu77, 2025-01-08:
How was the camera initialized without being plugged in? Do you have a demo example?

Sija, 2025-01-08:
Camera is initialized while it's being plugged. It's as easy as turning it off/unplugging it afterwards.