gphoto/libgphoto2 #1071: camlibs\src\library.c camera_wait_for_event函数分支PTP_VENDOR_SONY && PTP_OC_SONY_QX_GetAllDevicePropData GP_EVENT_FILE_ADDED事件 break 之后变为 GP_EVENT_TIMEOUT

ArvinSpace, 2025-01-03:
**Describe the bug**

文件：camlibs\src\library.c
函数：camera_wait_for_event
以下分支：
```
	if (	(params->deviceinfo.VendorExtensionID == PTP_VENDOR_SONY) &&
		ptp_operation_issupported(params, PTP_OC_SONY_QX_GetAllDevicePropData)
	) {
```

疑似有BUG的所在行：https://github.com/gphoto/libgphoto2/blob/1584f5334b40933fcc85b3145c0aa27b025c765a/camlibs/ptp2/library.c#L7191

疑问如下：
此处是否应该`return GP_OK;`并且释放对应内存资源，而不应该是`break;`，因为`break;`之后事件会从`GP_EVENT_FILE_ADDED`变为`GP_EVENT_TIMEOUT`，这是BUG吗？

**Name the camera**
Sony

**libgphoto2 and gphoto2 version**
2.5.31

**To Reproduce**
sdf

msmeissn, 2025-01-03:
yes, we should return GP_OK here. thanks!

closed 2025-01-03 by msmeissn

ArvinSpace, 2025-01-03:
good, thank you.