gphoto/libgphoto2 #1072: 可以支持多线程监听gp_camera_wait_for_event的事件吗

ArvinSpace, 2025-01-07:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
普通的监听`gp_camera_wait_for_event`的事件，然后拉取每一张照片数据保存，这样速度比较慢，每个事件完成需要2秒多（与照相机本身的处理进度条时间基本一致）

所以现在我想通过`pthread`或者其他库，使用多线程监听`gp_camera_wait_for_event`的事件，这样应该能更快速拉取多张照片的数据
但是我这么做了之后，返回错误码`-100`，不能正确处理事件
请问有什么方式支持多线程吗？

**Describe alternatives you've considered**

**Additional context**

msmeissn, 2025-01-07:
Multithreading is not easy. The problem is that the camera communication is single threaded by nature. So at most libgphoto2 could be optimized, but in the end the limiting factor is the camera request round trips.

ArvinSpace, 2025-01-08:
> Multithreading is not easy. The problem is that the camera communication is single threaded by nature. So at most libgphoto2 could be optimized, but in the end the limiting factor is the camera request round trips.

Okay, thank you for your reply.

closed 2025-01-09 by msmeissn