gphoto/libgphoto2 #1073: ILCE-7M3, ILCE-7M4, FX3 get iso, f-number, shutterspeed range error

xujinbook, 2025-01-08:
**Describe the bug**

        The issue we are currently encountering is the inaccurate retrieval of adjustable parameters (ISO, Iris, Shutter Speed) on Sony cameras.
                
                Description of inaccurate ISO retrieval:
                         We conducted the tests using the A7M3 in manual mode. The ISO range displayed by the A7M3 does not match the range obtained via gphoto2, as shown below:
                                 A7M3 displayed ISO range:
                                        AUTO 50 60 64 80 100 125 160 200 250 320 400 500 640 800 1000 1250
                                        1600   2000  2500  3200  4000 5000 6400 8000 10000  12800  16000  
                                        20000 25600  32000  40000 51200  64000 80000   102400  128000 
                                        160000  204800
                                 Gpohto2 displayed ISO range:
                                        AUTO 50 60 64 80 100 125 160 200 250 320 400 500 640 800 1000 1250
                                        1600   2000  2500  3200  4000 5000 6400 8000 10000  12800  16000  
                                        20000 25600  32000  40000 51200  64000 80000   102400  128000 
                                        160000  204800  256000 320000 409600
                                  We have found that the range obtained using gphoto2 is broader than the actual range displayed by the A7M3. We hope to obtain a range that is consistent with
                                  what is  shown on the A7M3. (The command used is gphoto2 --get-config iso.)

                 Description of inaccurate F-Number retrieval:                         
                         We observed that the aperture values displayed by the A7M3 differ at various focal lengths, but the range of aperture values obtained using gphoto2 remains unchanged.
                                  The A7M3 supports f-number values from f/3.5 to f/32 when using a lens with a focal length of 28mm；
                                  The A7M3 supports f-number values from f/5.6 to f/36 when using a lens with a focal length of 70mm；
                                
                                  However, the f-number values obtained using gphoto2 do not change with different focal lengths and always display a range of f/1 to f/64. 
                                  (The command used is gphoto2 --get-config f-number.)

                  Description of inaccurate ShutterSpeed retrieval:   
                          Similar to ISO and F-Number, the range of ShutterSpeed obtained using gphoto2 is broader than the range supported by the A7M3.

                 We urgently need your valuable advice to obtain the correct parameter ranges. Thank you in advance for Sony's assistance.

**Name the camera**
ILCE-7M3, ILCE-7M4, FX3

**libgphoto2 and gphoto2 version**
output of:  the lastest master branch

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

jufa, 2025-01-09:
https://github.com/gphoto/libgphoto2/pull/1067

nicustraton, 2025-01-11:
I can solve Sony, Nikon, Canon cameras firmware problems/issues after firmware update etc.

If you want to unlock region for your Sony FX3/FX30 camera or other model, in order to have latest firmware and PAL-NTSC, write me on whatsap +40727005523, country code Romania (+40) and number 727005523 or mail me: nicustraton.baron at gmail dot com.

You just have to connect the camera with USB cable in Image Transfer (MSC) mode to a windows PC/laptop with Windows OS and Intel CPU. I will unlock camera first and you pay after.

PS: if you have an older firmware version on camera and you know you have your camera from grey market If you want to update firmware to latest version, if you will update, you will lose PAL/NTSC so I recommend to let me unlock region first, update firmware to latest version and then unlock region again.

* this is a safest way becauseI will backup your camera original settings and firmware before any procedures.

xujinbook, 2025-01-14:
I can not get correct iso f-number shutterspeed by lastest commit

xujinbook, 2025-01-14:
我们希望能够通过 gphoto2 获取的 iso，f-number，shutterspeed 范围以及枚举的形式展示出来，并且展示出来的值能够与相机（Sony ILCE-7M3, ILCE-7M4, FX3等）中实际参数范围值保持一致； @jufa @rheimbuch @nicustraton