gphoto/libgphoto2 #1078: libgphoto2 dies when trying to process my phot files (iphone, fuji XV100)

ericaquaronne, 2025-01-17:
**Describe the bug**
on my machine since a recent update I get errors running the libs libgphoto2 (on various apps)
it dies simply :-(
I did a test to capture info using gPhoto2 via CLI

[my-logfile.txt](https://github.com/user-attachments/files/18457119/my-logfile.txt)

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
see attached log, is does not detect neither the iphone nor the Fuji XV100 files 

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2 2.5.15

Copyright (c) 2000-2017 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  -

[debug.log](https://github.com/user-attachments/files/18457083/debug.log)

   if considered useful.

msmeissn, 2025-01-19:
can you capture debug output of:

gphoto2 --summary --debug --debug-logfile=summary-debug.txt

and attach summary-debug.txt ?

ericaquaronne, 2025-01-20:
here it is

[summary-debug.txt](https://github.com/user-attachments/files/18477609/summary-debug.txt)