gphoto/libgphoto2 #1087: Newest Sony fixes break Nikon D800e (and probably others)

olvidalo, 2025-02-03:
**Describe the bug**

Since commit 152b228f99558f7586bdbc7545bad0a5242c96d2, libgphoto2 segfaults at `wait_for_event` most of the time when a setting is changed manually on the camera.  

I could trace it back to `dpd` being `NULL` in `ptp_unpack_DPD` `ptp-pack.c`, line [658](https://github.com/gphoto/libgphoto2/blob/5a3491fba2faf2c78f1e210424bdef9beaa105ce/camlibs/ptp2/ptp-pack.c#L658). 

Here's a stack trace:
<details><summary>Details</summary>
<pre>
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x40)
  * frame #0: 0x0000000100522dfc ptp2.so`ptp_unpack_DPD(params=0x000000015ae06fe0, data="\aP\U00000004", dpd=0x0000000000000000, dpdlen=56, offset=0x000000016fdfeab0) at ptp-pack.c:658:2 [opt]
    frame #1: 0x0000000100522904 ptp2.so`ptp_getdevicepropdesc(params=0x000000015ae06fe0, propcode=20487, devicepropertydesc=0x0000000000000000) at ptp.c:2761:8 [opt]
    frame #2: 0x000000010052bfa8 ptp2.so`ptp_generic_getdevicepropdesc(params=0x000000015ae06fe0, propcode=20487, dpd=0x000000016fdfee60) at ptp.c:4714:3 [opt]
    frame #3: 0x0000000100540024 ptp2.so`camera_wait_for_event(camera=<unavailable>, timeout=<unavailable>, eventtype=<unavailable>, eventdata=<unavailable>, context=<unavailable>) at library.c:7352:9 [opt]
    frame #4: 0x0000000100045f60 libgphoto2.6.dylib`gp_camera_wait_for_event(camera=0x000060000362c270, timeout=5000, eventtype=0x000000016fdff064, eventdata=0x000000016fdff058, context=0x0000600000a2c000) at gphoto2-camera.c:1446:2 [opt]
</pre>
</details> 

This goes back to [this change](https://github.com/gphoto/libgphoto2/blob/5a3491fba2faf2c78f1e210424bdef9beaa105ce/camlibs/ptp2/ptp.c) 152b228f99558f7586bdbc7545bad0a5242c96d2 introduced in 152b228f99558f7586bdbc7545bad0a5242c96d2 where `dpd_in_cache` is set to `NULL` under some circumstances

```c
	if (dpd_in_cache->DataType != PTP_DTC_UNDEF) {
		if (dpd_in_cache->timestamp + params->cachetime > now) {
			duplicate_DevicePropDesc(dpd_in_cache, dpd);
			return PTP_RC_OK;
		}
		/* free cached entry as we will refetch it. */
		ptp_free_devicepropdesc (dpd_in_cache);
		dpd_in_cache = NULL; // <--- HERE 
	}
```

which is then passed to `ptp_getdevicepropdesc` (and on to `ptp_unpack_DPD`) [here](https://github.com/gphoto/libgphoto2/blob/5a3491fba2faf2c78f1e210424bdef9beaa105ce/camlibs/ptp2/ptp.c#L4714):

```c
generic:
	if (ptp_operation_issupported(params, PTP_OC_GetDevicePropDesc)) {
		CHECK_PTP_RC(ptp_getdevicepropdesc (params, propcode, dpd_in_cache));
		goto done;
	}
```

**Possible fixes**
Removing `dpd_in_cache = NULL` fixes the error and makes the camera usable again, but I don't know what else it might break.

Another thing that worked is to re-fetch the DPD from the cache before passing it on, as in:

```c
generic:
	if (ptp_operation_issupported(params, PTP_OC_GetDevicePropDesc)) {
                 dpd_in_cache = ptp_find_dpd_in_cache(params, propcode);
		CHECK_PTP_RC(ptp_getdevicepropdesc (params, propcode, dpd_in_cache));
		goto done;
	}
```

But there might me a better way. As I'm new to this library and I don't *really* know what's going on in this code, I'd like to leave it to someone more familiar with the codebase to determine the proper fix. Once someone can point out the correct approach, I'm happy to submit a PR (if that would be helpful).

**Name the camera**
Nikon DSC D800E

**libgphoto2 version**
152b228f99558f7586bdbc7545bad0a5242c96d2 and newer

**To Reproduce**
- connect to the camera
- call `gp_camera_wait_for_event` in a loop
- change some setting manually on the camera (e.g. f-number)

Here's a small program I tested with: https://gist.github.com/olvidalo/3682c6a474400cb79f4cbd408213d4a0.
Example output: 
```
Listening for PTP events...
Unknown event received. Data: PTP Property 0000500d changed, "shutterspeed" to "0.0769s"
Unknown event received. Data: PTP Property 0000d100 changed, "shutterspeed2" to "1/13"
Unknown event received. Data: PTP Property 0000d1b1 changed, "lightmeter" to "-6.000000"
Unknown event received. Data: PTP Property 00005007 changed, "f-number" to "f/9"
zsh: segmentation fault
```

msmeissn, 2025-02-04:
i put in a fix, can you test?

 @watsonmw fyi regression fix for your change

olvidalo, 2025-02-04:
Tested it and it works, thank you!

msmeissn, 2025-02-04:
thanks for the feedback!

closed 2025-02-04 by msmeissn